UIPEthernet/UIPEthernet #103: UIP ethernet large

gmeijer, 2020-07-20:
To implement SSL, I'm trying to use the SSL client library: https://github.com/OPEnSLab-OSU/SSLClient
In this implementation they are using ethernetLarge where the number of sockets are reduced and a define ethernetLarge is added.
https://github.com/OPEnSLab-OSU/EthernetLarge
I'm not sure how this should be done in UIPethernet. Can someone advise?
Kind regards,
                      Gerard.

jandrassy, 2020-07-20:
the large buffer setting is specific for the W5x00 chips how they work internally. the count of connections is there in balance with this setting. the changes are not applicable to UIPEthernet library.

the count of connections is in utility/uipethernet-conf.h, but it will not make more space in the memory of the ENC26J60 where the network packets are.

gmeijer, 2020-07-21:
Thank you for your quick response. I managed to get the SSL running on the Arduino Due with UIPethernet.
Thanks for your help!

closed 2020-07-21 by gmeijer

xreef, 2022-03-04:
Hi @gmeijer,
Can you explain what chang did you made?
Thanks, Renzo

jandrassy, 2022-03-04:
> Hi @gmeijer, Can you explain what chang did you made? Thanks, Renzo

no change is needed. 
for Arduino I recommend the EthernetENC library

xreef, 2022-03-04:
I Try It and finally I can send email with Gmail (SSL) with my esp32 and enc28j60 (already tested with w5500 and EthernetLarge).
Now only in branch https://github.com/xreef/EMailSender/tree/features/SSLClientTest .
I'm moved!!

jandrassy, 2022-03-04:
for esp32 it is better to use the Ethernet peripheral of the MCU
https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino

xreef, 2022-03-07:
Hi @jandrassy,
Thanks for your integration, but that was only a test.
my target is to send SSL EMail with STM32 (with w5500 -> OK and enc28j60).
Thanks, Renzo