UIPEthernet/UIPEthernet #104: atmega freezes when tring to publish

testter2, 2020-07-22:
Using ATMEGA2560 with enc28j60 (ethernet module).

I'm using UIPEthernet library for enc28j60 module and pubsubclient library to publish in mqtt.

I'm trying to publish JSON data of large sizes about (200 to 300 bytes) and I used the next to the line to increase the buffer.
#define MQTT_MAX_PACKET_SIZE 1024

everything is going right and I can publish without any problems but after some time (which is random ) the atmega freezes.

I traced the freezing and it's always happening after this line of code:
_client->write(buf+(MQTT_MAX_HEADER_SIZE-hlen),length+hlen)

this's my publish function

bool publishMessage(String ABOUT_MESSAGE, String MESSAGE)
{

  Serial.print(F("Publishing ::: "));
  Serial.println(MESSAGE);

  if (mqttClient.state() == 0) // Check client connection
  {
    Serial.print(F("AFTER mqttClient.state() == 0 "));
    bool publishSuccess = mqttClient.publish("PostSensorData", MESSAGE.c_str());
    Serial.print(F("AFTER mqttClient.publish "));
    if (publishSuccess) // Check publish success
    {
      Serial.print(F("publish "));
      Serial.print(ABOUT_MESSAGE);
      Serial.println(F( " ok..."));
      return true;
    }
    else
    {

      Serial.print(F("publish "));
      Serial.print(ABOUT_MESSAGE);
      Serial.println(F( " failed..."));

      return false;
    }
  }
  else
  {
    Serial.print(F("Publish failed... MQTT client state is : "));
    Serial.println(mqttClient.state());

    return false;
  }
  return true;

}

is there any solution for this problem?

thanks in advance.

jandrassy, 2020-07-22:
https://majenko.co.uk/blog/evils-arduino-strings

closed 2020-07-22 by jandrassy

testter2, 2020-07-22:
> 
> 
> https://majenko.co.uk/blog/evils-arduino-strings

I think that isn't a memory leakage as I used the following function to check the free memory in the loop and the result always = 2527.

```
int freeRam () {
  extern int __heap_start, *__brkval;
  int v;
  return (int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval);
}
```
is there any other problems you think ?!

testter2, 2020-07-22:
> 
> 
> https://majenko.co.uk/blog/evils-arduino-strings

I think that isn't a memory leakage as I used the following function to check the free memory in the loop and the result always = 2527 besides the atmega hangs after random time.

```
int freeRam () {
  extern int __heap_start, *__brkval;
  int v;
  return (int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval);
}
```
could it be related to the router of the internet?

is there any other problems you think ?!