UIPEthernet/UIPEthernet #113: Board stops working after certain amount of time.

droided4u, 2020-08-29:
UNO + Enc28j60 stops working after 5-6 hours if I don't soft reset.
It's really no problem atm since my project is working fine and the soft reset is almost instantaneous so it doesn't really bother me all that much but if it's something I can fix i would like to know.

When it's stuck the lights are on but I get no connection over the internet/local.

Note: The UNO board is not genuine so that can be the causing issues as well.

Also sorry for lots of comments, It's because I posted this on the Arduino forum and I like helping people out as much as possible. After all, not everyone is after learning something, some are after getting something done and forget all about it.
I could not find any project involving enc28j60, UIPEthernet, and WOL together so I wanted to create one.

My Arduino and c++ knowledge is quite limited so if you find any other mistakes within the code please do warn me. I had to create my own magic packet because seemingly there is no such thing in the library.
I will convert those if statements to a function later, I know they look kind of foolish but I have only 1 computer that I wake so a function doesn't really benefit my case at all.

`#include <UIPEthernet.h>
 
byte mac[] = { 0x74,0x69,0x69,0x2D,0x30,0x31 }; //mac address for enc28j60 you change or keep does not matter.
byte ip[] = { 192, 168, 1, 177 }; // IP address of enc28j60, I use 192.168.1.177 internal IP, I also reserved this IP for MAC above from router so that no device takes its place
EthernetServer server(80); //server port 80 meaning HTTP
EthernetUDP udp;
String readString;

byte targetMac1[6] = {0x11,0x11,0x11,0x11,0x11,0x11}; //Target PC MAC 1 example MAC: 5C:D2:34:AE:7B:5D ==> {0x5C,0xD2,0x34,0xAE,0x7B,0x5D};
byte targetMac2[6] = {0x22,0x22,0x22,0x22,0x22,0x22}; //Target PC MAC 2 you can add more PC's here, if you don't have comment out
 

void setup(){
 

  Ethernet.begin(mac, ip);
  server.begin();

}
 
void loop(){
  // Create a client connection
  EthernetClient client = server.available();
  if (client) {
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
 
        //read char by char HTTP request
        if (readString.length() < 100) {
 
          //store characters to string
          readString += c;
        }
 
        //if HTTP request has ended
        if (c == '\n') {
/////////////////////////////////////////////////////////////////////
  // YOUR WEB PAGE IS HERE
          client.println("HTTP/1.1 200 OK"); //send new page
          client.println("Content-Type: text/html");
          client.println();
          client.println("<HTML>");
          client.println("<HEAD>");
          client.println("<TITLE>N'S REMOTE CONTROL</TITLE>"); // PAGE TITLE
          client.println("</HEAD>");
          client.println("<form>");
          client.println("<H1>Enter PC ID :");
          client.println("<input type=\"text\" name=\"text1\" style=\"font-size:18pt;\" >"); //create a text box to enter PC ID IMPORTANT TO UNDERSTAND, can be resized by adjusting px
          client.println("  <input type=\"button\" value=\"TURN ON\" name=\"Submit\" style=\"font-size:18pt;\" onclick=javascript:load(text1.value)></H1>"); // Button to redirect, can be resized by px
          client.println("</form>");
          client.println("<script language = \"javascript\">");
          client.println("function load(url)"); //function to redirect to PC ID url
          client.println("{");
          client.println(" location.href='?'+url;");
          client.println("}");
          client.println("</script>");
          client.println("</HTML>");
          delay(1);
          //stopping client
          client.stop();
// WEB PAGE ENDS
///////////////////////////////////////////////////////////////////////
         
          udp.beginPacket(IPAddress(255,255,255,255),9); // begin collecting UDP packet to be broadcasted 255.255.255.255 from port 9

////////////////////////////////////////////////////////////////////
	  if(readString.indexOf("?your-pc-id-link") >0)  // change your-pc-id-link, you will have to enter this link to chat box in html page, do not remove the "?" char. Example "?HP-notebook"
          {
           
            for (int i = 0; i < 6; i++){          // Write first 6 FF HEX values of magic packet.
              udp.write(0xFF);
            }

            for (int i = 0; i < 16; i++){         // Write targetMAC1 address for magic packet
              udp.write(targetMac1,6);
            }
          }
///////////////////////////////////////////////////////////////////
          else if(readString.indexOf("?your-second-pc-id") >0)  // For second PC if you don't have one comment out else if block for more RAM.
          {

            for (int i = 0; i < 6; i++){
            udp.write(0xFF);
              }

              for (int i = 0; i < 16; i++){
                udp.write(targetMac2,6);
              }
          }
///////////////////////////////////////////////////////////////////
          udp.endPacket();             
          udp.stop(); //package is sent

          //clearing string for next read
          readString="";
 
        }
      }
    }
  }
}`

droided4u, 2020-08-29:
I'm sorry idk how I can fix that html mess.

closed 2020-08-29 by droided4u

jandrassy, 2020-08-29:
it is always the String class
explanation and solution: https://majenko.co.uk/blog/evils-arduino-strings