UIPEthernet/UIPEthernet #115: Breakout problem with pubsubclient

nedimkivanc, 2020-08-31:
Hi,

I am using pubsubclient with uipethernet and enc28j60, esp32. The MQTT connection is in constant disconnection and data is constantly missing. I did the same test with Ardunio mega and the same problem exists with ardunio mega. I see some data coming in between, but only 1 out of 10 data sent

[image]

#include <UIPEthernet.h> 
#include <PubSubClient.h>

#define mqtt_server "*****"
#define mqtt_user "****"      // if exist
#define mqtt_password "***"  //idem

EthernetClient eth;
PubSubClient client(eth);

void setup() {
  byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
  Serial.begin(9600);
  Ethernet.begin(mac);
  Serial.println(Ethernet.localIP());
  Serial.print("subnetMask: ");
  Serial.println(Ethernet.subnetMask());
  Serial.print("gatewayIP: ");
  Serial.println(Ethernet.gatewayIP());
  Serial.print("dnsServerIP: ");
  Serial.println(Ethernet.dnsServerIP());
  client.setServer(mqtt_server, 15550);
}

void reconnect() {

  while (!client.connected()) {
    Serial.print("Connecting to MQTT broker ...");
    if (client.connect("ESP8266Client", mqtt_user, mqtt_password)) {
      Serial.println("OK");
      delay(4000);
    } else {
    Serial.print("KO, error : ");
    Serial.print(client.state());
    Serial.println(" Wait 4 secondes before to retry");
    delay(4000);
}
  }
}

void loop() {
   if (!client.connected()) {
  reconnect();
  }
delay(1000);
client.loop();
client.publish("esp.test", "123123123");
Serial.println("gönder");
Serial.print(client.state());
Serial.println("\n");
delay(5000);
}

jandrassy, 2020-08-31:
the GitHub version of the library?

nedimkivanc, 2020-08-31:
yes, ı used GitHub Version

jandrassy, 2020-08-31:
> yes, ı used GitHub Version

downloaded later then 23th august?

nedimkivanc, 2020-08-31:
yes i downloaded the library today. Same situation when I change the delays

jandrassy, 2020-08-31:
> yes i downloaded the library today. Same situation when I change the delays

remove the delays and use timing with millis. call Ethernet.maintain() in every loop.
or at least replace the delays with
```
for (int i =0; i < 100; i++) {
  delay(50); // or 30 or 20
  Ethernet.maintain();
}
```

nedimkivanc, 2020-08-31:
Thanks for help, It's really great work. I hope it helps other people too

closed 2020-08-31 by nedimkivanc