UIPEthernet/UIPEthernet #120: hardwareStatus() not working

dioannidis, 2020-09-25:
Hi,

   AFAIU, UIPEthernetClass::hardwareStatus() is implemented querying the erevid variable from Enc28J60Network.cpp which IMHO is not logically correct.

   The erevid variable, which is assigned 0 at definition, is populated in the Enc28J60Network::init method, which is called from the UIPEthernet::netInit() method, which is called from the UIPEthernetClass::begin() method.

   AFAIU, the hardwareStatus method use case, is to "see" ahead the existence of the chip and then call UIPEthernetClass::begin(). But until someone call UIPEthernetClass::begin(), the erevid variable is 0, which effectively makes the call to UIPEthernetClass::hardwareStatus() useless. It always return EthernetNoHardware.

regards,

jandrassy, 2020-09-25:
how would you do it?
int() sets the CS pin, begin() initializes SPI. only then we can talk to ENC

dioannidis, 2020-09-25:
Hi,

  ( brainstorming mode ) 

 One approach, maybe, is 

- introduce a new define for the CS pin. A define that a user can override in his / her code ( not in utility dir ) . 

- move SPI initialization from to be done on library load. At init() leave only the enc configuration 

What do you think ?

regards,

dioannidis, 2020-09-25:
Hi,

  ( brainstorming mode continues )

  also with the above structure we can avoid the SPI initialization on library load if the library tracks SPI initialization. Maybe checking if erevid = 0 ( to not introduce another variable ), the SPI initialization to avoid reinitialization from hardwareStatus() and begin() calls.

  If hardwareStatus() method called first, initializes the SPI and retrieve the erevid. The begin() method ( when called ) checks if the erevid <> 0, to avoid to initialize the SPI again. If the begin() method, checks the erevid and is equal to 0, then the begin() method initializes the SPI, assuming that there wasn't a prior hardwareStatus() call.

regards,

jandrassy, 2020-09-25:
but you see that it is too much work for too less benefit?

dioannidis, 2020-09-25:
Hi,

  well, I don't know if it's too much work, I'll take a look more closely and get back.

  My motivation for this issue was that I tried to use hardwareStatus() method and I wrongly assumed that the enc28j60 module was faulty. Currently it gives faulty results.

  regards,

jandrassy, 2020-09-25:
I made in August a new version of UIPEthernet called EthernetENC. I found this issue as I tested the Ethernet library's LinkStatus example. But I only put a note in Examples page in Wiki.

dioannidis, 2020-09-26:
Hi,

  please look at my fork ( https://github.com/dioannidis/UIPEthernet/commit/303d775418fa509551862e5702c595fa5c53a79a ) for a possible solution.

regards,

jandrassy, 2020-09-26:
yes. how did the flash size usage change?

dioannidis, 2020-09-26:
Hi,
  
  what is the "proper" acceptable procedure to measure the flash size ?

regards,

jandrassy, 2020-09-26:
Arduino IDE
Sketch uses ...
Global variables use ...

dioannidis, 2020-09-26:
Hi,

  empty sketch with only include UIPEthernet ? Or some specific sketch ?

regards,

dioannidis, 2020-09-26:
Hi,

  anyway, using the following sketch
 
`#include <UIPEthernet.h>

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED }; 

void setup() {
  Ethernet.hardwareStatus();
  Ethernet.begin(mac);
}

void loop() {
  // put your main code here, to run repeatedly:

}`

I'm getting from 2.0.9 lib 
"Sketch uses 16758 bytes (51%) of program storage space. Maximum is 32256 bytes.
Global variables use 979 bytes (47%) of dynamic memory, leaving 1069 bytes for local variables. Maximum is 2048 bytes."

and after my changes
"Sketch uses 16938 bytes (52%) of program storage space. Maximum is 32256 bytes.
Global variables use 980 bytes (47%) of dynamic memory, leaving 1068 bytes for local variables. Maximum is 2048 bytes."

regards,

jandrassy, 2020-09-26:
180 bytes of flash? try to remove geterevidSPI(). revid was read once at the end of init(). now it could be read at the end of initSPI().

dioannidis, 2020-09-26:
Hi,

  this results is with the removal of geterevidSPI() method. I already show this and done the removal .

regards,

jandrassy, 2020-09-26:
use `bool Enc28J60Network::spiInitialized = false;`
the code style in this repo is
```
if (spiInitialized)
  return;
```

closed 2020-10-01 by jandrassy