UIPEthernet/UIPEthernet #124: UDP ignoring packets?

pascoea, 2020-10-05:
Hey there.  I've been messing with this for a couple days now and can't figure it out.
I'm attempting to use the library to receive/process UDP packets for a lighting controller (e1.31 packets generated by Vixen Lights)

Board is an ESP8266E (Heltec Wifi Kit 8)

Vixen generates a packet for each Universe in the e1.31 configuration, in this instance there are 4 universes.  The sketch processes packets 1 and 2, ignores 3 and 4, then repeats.

I've troubleshot it down to line 374 in UIPUdp.cpp:  if (data->packet_next == NOBLOCK)  For the packets it skips, this test fails, so the uipudp_appcall method essentially ends.

jandrassy, 2020-10-06:
yes the library stores only two message at once for one port.
I guess if you remove the logging you can read the messages fast enough to process a message before the next arrives.

closed 2020-10-06 by jandrassy

pascoea, 2020-10-06:
Thank you for the reply.  Even with having the program do nothing but grab the packet, print a couple characters out of serial, and return to the loop it misses packets.  I'll have to do some more experimentation.

Is the 2 packet a limitation/feature of the uIP library, or the drivers/wrappers?

jandrassy, 2020-10-07:
the Serial printing is very very slow compared to speed of network, but the MCU is faster then the network. 

the limitation is in UIPUdp. packet_in is message to read, packet_next is one waiting message

jandrassy, 2020-10-07:
I decided to implement UDP backlog with configurable size in my EthernetENC library

pascoea, 2020-10-08:
Awesome, look forward to seeing it!  Shoot me a note if there is anything I can do to help.

jandrassy, 2020-10-08:
but for esp8266 maybe there is a better solution. in esp8266 arduino 3.0, enc28j60 will be supported as additional interface for the ESP8266WiFi library. now it has two interfaces STA and SoftAP. then it will support W5500, enc28j60, PPP...

you could try it https://github.com/esp8266/Arduino/pull/6680

jandrassy, 2020-10-10:
> I decided to implement UDP backlog with configurable size in my EthernetENC library

done
https://github.com/jandrassy/EthernetENC

pascoea, 2020-10-14:
Thank you kindly! This (mostly) solved the problem.  I still have some exploring to do, but I was to accomplish what I wanted to.

Note to anyone that finds this:
One of my struggles was an extremely limited receive buffer.  In the enc28j60.h file, the line #define RXSTOP_INIT (0x1FFF-0x1800) only allocates 2k of incoming buffer.  I updated it to 0x1FFF-0x0D00 (~4.75k incoming), this may still not be optimal, just where I arrived at via testing.

jandrassy, 2020-10-14:
the library moves data from RX packets, to 'TX' area to store them until they are processed by application.
do you have some delays or other code which takes time and doesn't call any function of UIPEthernet? because then the RX queue is not handled. Call some function on some object of UIPEthernet library at least every 250 milliseconds.  Call Ethernet.maintain() if you can't call parsePacket() so frequent.