UIPEthernet/UIPEthernet #126: Get the link status of the enc28j60

ColinDexter, 2020-11-27:
How can I get the link status of the enc28j60. On the library of Nicholas Humfrey (https://github.com/jcw/ethercard/) I use ether.isLinkUp()

But that is not working in this library

jandrassy, 2020-11-27:
https://www.arduino.cc/en/Reference/EthernetLinkStatus

for new projects it is better to use EthernetENC library

closed 2020-11-27 by jandrassy

ColinDexter, 2020-11-27:
Thanks, but that one is not working on an esp8266. It keeps restarting 🙁

https://www.arduino.cc/en/Reference/EthernetLinkStatus

jandrassy, 2020-11-27:
> 
> Thanks, but that one is not working on an esp8266. It keeps restarting 🙁

EthernetENC? did you set the CS pin with Ethernet.init?

ColinDexter, 2020-11-27:
I don't see anything about setting the CS pin with Ethernet.init? It only says "Only change #include <Ethernet.h> to #include <EthernetENC.h"

jandrassy, 2020-11-27:
you would have to set the CS pin with a W5500 module too
https://www.arduino.cc/en/Reference/EthernetInit
only UIPEthernet tries to guess the CS pin for esp

ColinDexter, 2020-11-27:
Thank you for your answer. Then I'm probably doing something wrong. It just keeps restarting with me :-(
I have Ethernet.init (15); because I think this is GPIO15 that CS is connected to. I am using the following hardware https://github.com/UIPEthernet/UIPEthernet/blob/master/hardware/NodeMCU_enc28j60_wiring.PNG. And the following code to test.

#include <SPI.h>
#include <EthernetENC.h>

byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};

void setup() {
  Serial.begin(9600);
  Ethernet.init(15);

  Ethernet.begin(mac);
  Serial.print("localIP: ");
  Serial.println(Ethernet.localIP());
  Serial.print("subnetMask: ");
  Serial.println(Ethernet.subnetMask());
  Serial.print("gatewayIP: ");
  Serial.println(Ethernet.gatewayIP());
  Serial.print("dhcpServerIP: ");
  Serial.println(Ethernet.dnsServerIP());
  Serial.print("dnsServerIP: ");
  Serial.println(Ethernet.dnsServerIP());
}

void loop() {
}

https://www.arduino.cc/en/Reference/EthernetInit

jandrassy, 2020-11-27:
you made that circuit for D8? remove it and use some other other pin as CS. for example D1.
btw, you don't have to guess the pin number Arduino core knows D8 if you select NodeMcu in Tools menu

ColinDexter, 2020-11-28:
No, unfortunately another pin for CS does not work either. And when I use the UIPEthernet library it works all at once. 🙁

Will see if there is another library somewhere.

jandrassy, 2020-11-28:
> Will see if there is another library somewhere.

why? UIPEthernet has linkStatus too

how many Arduino libraries for enc28j60 you think exists? UIPEthernet and EthernetENC are based on the same older library. 

the truth is ESP32 Arduino core could support the enc28j60 as an additional interface with the LWIP TCP/IP stack used for the WiFi. but it doesn't. it only supports the Ethernet peripheral of ESP32.

jandrassy, 2020-11-29:
sorry, I confused two troubleshooting sessions. the other had esp32.
the esp8266 Arduino core 3.0 will support enc28j60 as additional  networking interface.
you can test it https://github.com/esp8266/Arduino/pull/6680

ColinDexter, 2020-11-29:
> how many Arduino libraries for enc28j60 you think exists?

I know, very little...

ColinDexter, 2020-11-29:
Can happen, no problem

I think my esp is not functioning properly. One time everything wakes up and works fine, the next time when I recompile everything it will continuously restart. If I add Ethernet.linkStatus () to the code it not works at all, it then reboot continuously.

So I think my esp is not quite good. In addition, I was looking to get more information from the DHCP. Such as requesting the DHCP server ip address, time server and boot file. But the whole thing is so unstable that this doesn't really work. So I will first order a new esp8266 ...

you can test it esp8266/Arduino#6680<https://github.com/esp8266/Arduino/pull/6680>

jandrassy, 2020-11-29:
try to clear the flash memory with esptool. it will clear the RF calibration data which could be wrong from using a different version of SDK or a different flash partitioning. additionally, the enc28j60 has a high current consumption too so you could try to turn off WiFi. see the EarlyDisableWiFi example

ColinDexter, 2020-11-29:
Thanks, I'm going to try this.

ColinDexter, 2020-12-13:
Hi Juraj,

I have a new esp8266 and it works fine now. Another question, do you maybe know how to request the domainName, dhcpIPaddrLeaseTime from the dhcp server and the IP address of the dhcp server it self? I can't get it done. And cannot find an example. I see domainName and dhcpIPaddrLeaseTime are mention in enum on Dhcp.h of the UIPEthernet.

Regards,
Colin

jandrassy, 2020-12-13:
Dhcp.h and Dhcp.cpp are there only for Ethernet.begin(mac). This is copied from Arduino Ethernet library.
The UIPEthernetClass doesn't expose the dhcp object which stores the DHCP parameters. The DHCP class exposes only parameters needed for Ethernet.begin(). 
So to access DHCP parameters you have to add accessors to UIPEthernetClass class and to DhcpClass

ColinDexter, 2020-12-14:
ah, okay I see thanks. Only get the necessary information unfortunately still not yet out. Will try it again tomorrow. Also for dns I only get one IP address while the dhcp gives two.

Regards,
Colin

colin.dexter@outlook.com

ColinDexter, 2020-12-19:
Hi,

When I use UIPEthernet with a wemos (esp8266) it works perfectly. But because you indicated that I can make better use of EthernetENC for new projects, I also installed this.

But almost immediately after Ethernet.begin(mac) I get a Soft WDT reset. When I look for this it is because somewhere a part of code running for over 2.5secs without a yield () having been executed. delay () executed yield (), and an automatic yield () is executed at the end of loop ().

Do you have any idea where this is in the code could be? I already fed yield () here and there. But without any results.

Best regards,
Colin

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffc0 offset: 01a0
3ffffec0:  00000fa0 3ffe84d3 00000060 0000ea60
3ffffed0:  40100274 00000100 3ffeed38 40209608
3ffffee0:  00000fa0 3ffeed38 00000100 40209bbd
3ffffef0:  00000fa0 3ffeed38 3ffe84cc 40209d5c
3fffff00:  3ffe8693 00000001 00203a6b 0000ea60
3fffff10:  00000fa0 3ffe84cc 3ffe84cc 40208413
3fffff20:  4020b38c 3ffeed58 3ffeebac 40208548
3fffff30:  4020b38c 3ffeed58 3ffe8691 4020b6dd
3fffff40:  3fffdad0 00000006 3ffeed58 4020b7d4
3fffff50:  3fffdad0 3ffeed58 3ffeed58 00000001
3fffff60:  3fffdad0 3ffeed58 3ffeeb98 4020557d
3fffff70:  402169c8 61747300 2e737574 8000736a
3fffff80:  3fff0ce0 feefeffe feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3ffeedd4
3fffffa0:  3fffdad0 00000000 3ffeed94 4020ce50
3fffffb0:  feefeffe feefeffe 3ffe84ec 40100c09
<<<stack<<<

jandrassy, 2020-12-19:
could you please decode the stack dump?

ColinDexter, 2020-12-19:
Decoding stack results
0x4010029d: __digitalWrite(uint8_t, uint8_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 86
0x40209464: Enc28J60Network::readOp(unsigned char, unsigned char) at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\utility\Enc28J60Network.cpp line 423
0x40209618: Enc28J60Network::readReg(unsigned char) at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\utility\Enc28J60Network.cpp line 497
0x40209a2c: Enc28J60Network::sendPacket(unsigned char) at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\utility\Enc28J60Network.cpp line 260
0x402086b4: uip_ipchksum() at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\Ethernet.cpp line 472
0x402082b4: UIPEthernetClass::network_send() at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\Ethernet.cpp line 302
0x4020919a: EthernetUDP::_send(uip_udp_conn*) at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\EthernetUdp.cpp line 387
0x40209205: EthernetUDP::endPacket() at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\EthernetUdp.cpp line 177
0x4020a2f4: DhcpClass::send_DHCP_MESSAGE(unsigned char, unsigned short) at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\Dhcp.cpp line 248
0x4010029d: __digitalWrite(uint8_t, uint8_t) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 86
0x4020a6d1: DhcpClass::request_DHCP_lease() at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\Dhcp.cpp line 62
0x40209618: Enc28J60Network::readReg(unsigned char) at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\utility\Enc28J60Network.cpp line 497
0x40209c68: Enc28J60Network::getrev() at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\utility\Enc28J60Network.cpp line 558
0x4020a828: DhcpClass::beginWithDHCP(unsigned char*, unsigned long, unsigned long) at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\Dhcp.cpp line 24
0x40208564: UIPEthernetClass::begin(unsigned char const*, unsigned long, unsigned long) at C:\Users\user\Documents\Arduino\libraries\EthernetENC-master\src\Ethernet.cpp line 67
0x4020b398: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266/HardwareSerial.h line 164
0x4020b6e9: Print::write(char const*) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266/Print.h line 62
0x4020b7e0: Print::println() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\Print.cpp line 186
0x4020558d: loop() at D:\Data\arduino\test/test.ino line 176
0x4020ce5c: loop_wrapper() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 197

jandrassy, 2020-12-19:
could you please use the GitHub master version of the library. there are some changes and the lines in stack trace will match the latest source code.
for now it looks like the sending of the DHCP request failed, which is the first thing to be send over network.

ColinDexter, 2020-12-19:
I have the same problem. I downloaded the library way from here: https://github.com/jandrassy/EthernetENC

If I remove Ethernet.init (D8); Ethernet.linkStatus() works. Only I don't get DHCP (Failed to configure Ethernet using DHCP)

if (Ethernet.begin(mac) == 0) {
  Serial.println("Failed to configure Ethernet using DHCP");
} else {
...
}

jandrassy, 2020-12-19:
try to use D1 as CS. D8 has a pull down resistor

ColinDexter, 2020-12-20:
When I do the init with D1 then I don't get a Soft WDT reset anymore. But also Ethernet.linkStatus () no longer works 
I have ordered also a NodeMCU board but I still waiting for it. When I have it I will test it with this board.

I would like to check the content of the packets. For that I would have to do the following:
byte Ethernet :: buffer [500];
ether.begin (sizeof Ethernet :: buffer, mac);
int plen = ether.packetReceive ();
if (byte_array_contains(Ethernet::buffer, 0, my_mac, sizeof(my_mac))) {
…
}

Maybe you know how I can solve that with EthernetENC or UIPEthernet library

jandrassy, 2020-12-20:
UIPEthernet and EthernetENC configure the ENC28J60 to filter the packets on MAC address so the ENC28J60 doesn't even put other packets in RX queue. it is because there is only 2kB for the RX queue to have space to store the packages that should be read by the sketch. the library was designed for small MCUs

ColinDexter, 2020-12-21:
That would be enough for me. I know the mac address from which I want to see the packets. Do you  have maybe an example of how you should do that? Something that’s prints out the packets of that specified given mac?

jandrassy, 2020-12-21:
what for what it be good? the purpose of the library is to implement TCP and UDP protocol over IP protocol. The ENC handles the raw packets. so to read them you only need something like the Enc28J60NetworkClass in utility/Enc28J60Network.h and .cpp

ColinDexter, 2020-12-22:
Ok, you can do that that way too. I did not know that. I'm not that good at programming. I will try to see if that works.

ColinDexter, 2021-01-08:
Hi Juraj,

Sorry to bother you one more time. Do you also know if UIPEthernet.h works together with an esp32. I try to use it with a LOLIN D32 but I don’t get it to work. I also tried to added port define as below in the code but it doesn't help.

    #define SPI_MOSI GPIO23
    #define SPI_MISO GPIO19
    #define SPI_SCK  GPIO18
    #define SPI_CS   GPIO15

Now I wonder if it works at all and if I can connect it directly or if I should use the transistor circuit with the BS170 again. I have tested both options. But none works 🙁.

So I hope you know if it works and how I should wire it.

Regards,
Colin

jandrassy, 2021-01-08:
no additional circuit. pin 15 isn't a bootconfig pin on esp32 and isn't a SS pin either. (but you can use it as CS). even on esp8266 you can use any other pin as CS a the "hold the pin HIGH" circuit is not necessary then.

for new projects use my EthernetENC library

ColinDexter, 2021-01-09:
Thanks for the info. It now works with a DOIT ESP32 DEVKIT V1. But when I try with a LOLIN D32 ethernet crashed after getting an IP address from the DHCP. On the moment when I have the IP address the link drops down and it stops working.

And unfortunately I still haven't working the EthernetENC library. It also not works with the esp32. I Have the same problem with it as with the esp8266...

Regards,
Colin

ColinDexter, 2021-01-11:
Hi Juraj,

I have been testing with the EthernetENC library for a while. I tested it with an arduino board, esp8266 and an esp32. With all three I can see the linkStatus but I am not getting an IP from the DHCP server. Even if I enter a fixed IP address, I cannot connect to the network. I don't understand why it works with UIPEthernet and why not with EthernetENC. Below the simple program I use to test.

#include <Arduino.h>
#include <EthernetENC.h>

uint8_t mac[6] = {0x00,0x01,0x02,0x03,0x04,0x05};

void setup() {
  Serial.begin(9600);

  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
  } else {
    Serial.print("localIP: ");
    Serial.println(Ethernet.localIP());
    Serial.print("subnetMask: ");
    Serial.println(Ethernet.subnetMask());
    Serial.print("gatewayIP: ");
    Serial.println(Ethernet.gatewayIP());
  }
}

void loop() {
}

What I also noticed is that Enc28J60Network::powerOff() and Enc28J60Network::powerOn() are not working. I am using the latest version EthernetENC downloaded from github.

And is there an example of how to do an ICMP ping. Can't find this anywhere. Or should I ask these questions on github.

Best regards,
Colin

jandrassy, 2021-01-11:
the UIPEthernet has macros to guess the CS pin. for EthernetENC you have to set CS pin with Ethernet.init.

ping (sending ICMP) is not supported in µIP