UIPEthernet/UIPEthernet #127: Tcp retransmission issue

MonirA1990, 2020-12-17:
The issue of tcp retransmission still exist with server when using webclient example, what should i do to solve it?
Thanks

jandrassy, 2020-12-17:
do you use a recent version of UIPEthernet? did you try EthernetENC library?
do you have a Wireshark log?

MonirA1990, 2020-12-17:
Yes I have recent version.
wireshark log:

31	17.395910	192.168.1.3	192.168.1.8	TCP	60	1026 → 80 [SYN] Seq=0 Win=512 Len=0 MSS=512
40	19.081484	192.168.1.3	192.168.1.8	TCP	60	[TCP Retransmission] 1026 → 80 [SYN] Seq=0 Win=512 Len=0 MSS=512
41	22.408640	192.168.1.3	192.168.1.8	TCP	60	[TCP Retransmission] 1026 → 80 [SYN] Seq=0 Win=512 Len=0 MSS=512
49	28.655850	192.168.1.3	192.168.1.8	TCP	60	[TCP Retransmission] 1026 → 80 [SYN] Seq=0 Win=512 Len=0 MSS=512
67	40.840658	192.168.1.3	192.168.1.8	TCP	60	1026 → 80 [RST, ACK] Seq=0 Ack=1 Win=512 Len=0

**This problem randomly occur not always.**
Thanks

MonirA1990, 2020-12-17:
Also if I try to connect with www.google.com, sometimes connection failed.

Initialize Ethernet with DHCP:
  DHCP assigned IP 192.168.1.3
connecting to www.google.com...
connection failed

disconnecting.
Received 0 bytes in 0.0000, rate = 0.00 kbytes/second

MonirA1990, 2020-12-17:
Here is another log:
29332	819.212896	192.168.1.9	192.168.1.8	TCP	60	[TCP Port numbers reused] 1026 → 8080 [SYN] Seq=0 Win=512 Len=0 MSS=512
29333	819.213093	192.168.1.8	192.168.1.9	TCP	58	8080 → 1026 [SYN, ACK] Seq=0 Ack=1 Win=65392 Len=0 MSS=1460
29342	820.213826	192.168.1.8	192.168.1.9	TCP	58	[TCP Retransmission] 8080 → 1026 [SYN, ACK] Seq=0 Ack=1 Win=65392 Len=0 MSS=1460
29349	820.963172	192.168.1.9	192.168.1.8	TCP	60	[TCP Retransmission] 1026 → 8080 [SYN] Seq=0 Win=512 Len=0 MSS=512
29350	822.216510	192.168.1.8	192.168.1.9	TCP	58	[TCP Retransmission] 8080 → 1026 [SYN, ACK] Seq=0 Ack=1 Win=65392 Len=0 MSS=1460
29356	824.211855	192.168.1.9	192.168.1.8	TCP	60	[TCP Retransmission] 1026 → 8080 [SYN] Seq=0 Win=512 Len=0 MSS=512
29371	826.215955	192.168.1.8	192.168.1.9	TCP	58	[TCP Retransmission] 8080 → 1026 [SYN, ACK] Seq=0 Ack=1 Win=65392 Len=0 MSS=1460
29373	830.460283	192.168.1.9	192.168.1.8	TCP	60	[TCP Retransmission] 1026 → 8080 [SYN] Seq=0 Win=512 Len=0 MSS=512
29408	842.756504	192.168.1.9	192.168.1.8	TCP	60	1026 → 8080 [RST, ACK] Seq=0 Ack=4264638797 Win=512 Len=0

jandrassy, 2020-12-18:
it is strange that uIP always sends port 1026. I doubt it is the right port of the connection.
I will try it later today. Which STM32 core do you use?

jandrassy, 2020-12-18:
sorry I can't test it now. I have the 3.3 V enc28j60 module in a long term stability test of EthernetENC library (6 days now) and the other module I have has a logic level conversion to 5 V so I can't wire it to Blue Pill SPI pins because these are not 5 V tolerant.

MonirA1990, 2020-12-19:
Thanks for you
Blue Pill SPI is 3.3v and do not need level conversion

jandrassy, 2020-12-19:
could you show your sketch or some test sketch with the same problem?

MonirA1990, 2020-12-19:
Here is the code:

/*
  Web client

 This sketch connects to a website (http://www.google.com)
 using an Arduino Wiznet Ethernet shield.

 Circuit:
 * Ethernet shield attached to pins 10, 11, 12, 13

 created 18 Dec 2009
 by David A. Mellis
 modified 9 Apr 2012
 by Tom Igoe, based on work by Adrian McEwen

 */

#include <SPI.h>
#include <UIPEthernet.h>

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

// if you don't want to use DNS (and reduce your sketch size)
// use the numeric IP instead of the name for the server:
IPAddress server(192,168,1,8);  // numeric IP for Google (no DNS)
//char server[] = "www.google.com";    // name address for Google (using DNS)

// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 1, 1);
IPAddress myDns(192, 168, 1, 1);

// Initialize the Ethernet client library
// with the IP address and port of the server
// that you want to connect to (port 80 is default for HTTP):
EthernetClient client;

// Variables to measure the speed
unsigned long beginMicros, endMicros;
unsigned long byteCount = 0;
bool printWebData = true;  // set to false for better speed measurement

void setup() {
  // You can use Ethernet.init(pin) to configure the CS pin
  //Ethernet.init(10);  // Most Arduino shields
  //Ethernet.init(5);   // MKR ETH shield
  //Ethernet.init(0);   // Teensy 2.0
  //Ethernet.init(20);  // Teensy++ 2.0
  //Ethernet.init(15);  // ESP8266 with Adafruit Featherwing Ethernet
  //Ethernet.init(33);  // ESP32 with Adafruit Featherwing Ethernet

  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // start the Ethernet connection:
  Serial.println("Initialize Ethernet with DHCP:");
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
      }
    }
    if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }
    // try to congifure using IP address instead of DHCP:
    Ethernet.begin(mac, ip, myDns);
  } else {
    Serial.print("  DHCP assigned IP ");
    Serial.println(Ethernet.localIP());
  }
  // give the Ethernet shield a second to initialize:
  delay(1000);
  Serial.print("connecting to ");
  Serial.print(server);
  Serial.println("...");

  // if you get a connection, report back via serial:
  if (client.connect(server, 80)) {
    Serial.print("connected to ");
    Serial.println(client.remoteIP());
    // Make a HTTP request:
    client.println("GET /index.php HTTP/1.1");
    client.println("Host: 192.168.1.8");
    client.println("Connection: close");
    client.println();
  } else {
    // if you didn't get a connection to the server:
    Serial.println("connection failed");
  }
  beginMicros = micros();
}

void loop() {
  // if there are incoming bytes available
  // from the server, read them and print them:
  int len = client.available();
  if (len > 0) {
    byte buffer[500];
    if (len > 500) len = 500;
    client.read(buffer, len);
    if (printWebData) {
      Serial.write(buffer, len); // show in the serial monitor (slows some boards)
    }
    byteCount = byteCount + len;
  }

  // if the server's disconnected, stop the client:
  if (!client.connected()) {
    endMicros = micros();
    Serial.println();
    Serial.println("disconnecting.");
    client.stop();
    Serial.print("Received ");
    Serial.print(byteCount);
    Serial.print(" bytes in ");
    float seconds = (float)(endMicros - beginMicros) / 1000000.0;
    Serial.print(seconds, 4);
    float rate = (float)byteCount / seconds / 1000.0;
    Serial.print(", rate = ");
    Serial.print(rate);
    Serial.print(" kbytes/second");
    Serial.println();

    // do nothing forevermore:
    while (true) {
      delay(1);
    }
  }
}

Thanks

MonirA1990, 2020-12-19:
The problem when trying to connect to server in client.connect(server, 80)
I try to change port of server but same behavior

jandrassy, 2020-12-20:
please try the EthernetENC library

and use
```
len = client.read(buffer, 500);
```

jandrassy, 2021-01-01:
so I tested it. it is only a problem of Wireshark. reset Wireshark session too.

closed 2021-01-01 by jandrassy