UIPEthernet/UIPEthernet #129: Network connection goes deaf.

pvrfr, 2021-01-29:
In this application, (https://opensprinkler.com/ an open source software and hardware lawn sprinkler controller), the system may run for a month but will then go deaf.   It will neither receive nor send packets, it will not respond to a ping.   The application code will run as normal doing other work but will not receive packets.   The system is not hung or crashed.   The network traffic is normally very light.  The traffic load does not seem to effect how often the problem happens.

I have compiled the entire OpenSprinkler system and added debug code.   The function uip_process in utility/uip.c is called repeatedly but does not return a packet.  The read call does not return a non-zero value.  I have a lot of debug information but the code flow through this function is quite convoluted.

This problem has been experienced by a number of users of the OpenSprinkler controller.   I am able to recreate this problem but it takes weeks of running.  Any help would be greatly appreciated.

jandrassy, 2021-01-29:
in EthernetENC library, which is is based on the same old library I know about a memory leak in enc28j60 RAM management. I don't know yet if it  is in the code which is common for both libraries or in code specific for the EtherneENC library.
I already fixed a ton of bugs in UIPEthernet and after every fix I thought it must be the last, there can't be so many.

pvrfr, 2021-01-29:
I had thought about a memory leak.   I put in some test code to check.   I did 2 tests, one for unbalanced stack
and the other for the heap.   In my debug print routine a printed the address of a stack variable.   If the stack
was unbalanced the address would have changed, but it did not.

To test the heap I put a malloc and a free together in my debug routing.   I printed the address of the allocated
memory.   If there was a leak the address should have changed, but it did not.

With those 2 tests I don't think that it is a memory leak.

I tried to setup the GDB stub driver into the code but it did not work properly for me.   There are people on the
net that claim to have GDB working on the ESP8266 board that is used in the OpenSprinkler.

Since my debug cycle is so long I have to carefully consider how I debug this thing.   I have an ASYNC port that
I can print debug messages through.   There are 3 buttons and a small display that I can use for debugging.

I have programmed one button to print a bunch of state information out the ASYNC port.   I have put many counters
in various places the uip_process routine.   This allows me to see where the code flows when the problem hits.
I can see that all the incoming packets are being dropped as the routine exits.   I don't yet have code to see the
contents of the packets.   Perhaps I will put that in for the next run.   Then I will have to wait a month.

Do you have any suggestion as to where to look?   Or suggestions for debugging the code?

Thanks,

pvrfr, 2021-01-30:
During the test runs I also recorded some of the states of the registers in the ENC28J60 chip.   During normal running these registers appear in one of 2 possible states:
estat 1 eir 28, econ1 4, epktcnt 0
estat 81 eir 68, econ1 4, epktcnt 1

The difference is random and just depends on when the state is captured.

When the network interface goes deaf this is the state of the registers.
estat 41 eir 29, econ1 4, epktcnt 0

The estat 0x40, bit 6, indicates a buffer error.
The eir 0x01, bit 0, indicates receive error interrupt.

It seems likely that this was caused by an incoming packet not being serviced in time.   Since the net load is very light it is unlikely that this is caused by CPU performance overload.

I suspect that there is some race condition where the incoming packet is not detected.

jandrassy, 2021-01-30:
the leak in EthernetENC is in enc28j60 RAM managed by the mempool.cpp. received net packages and packages to send are stored there.

I added `block->millis = millis();` in `MemoryPool::allocBlock` and can dump the mempool with
```
void MemoryPool::report(Print& out) {

  uint8_t count = 0;
  uint16_t size = 0;
  memblock *b = &blocks[POOLSTART];
  do
    {
      if (b->size > 0) {
        count++;
        size += b->size;
      }
      memhandle next = b->nextblock;
      if (next == NOBLOCK)
        break;
      b = &blocks[next];
      out.print("block ");
      out.print(next);
      out.print(" time ");
      out.print(b->millis);
      out.print(" address ");
      out.print(b->begin);
      out.print(" size ");
      out.print(b->size);
      out.println();
    }
  while (true);
  out.print("free blocks count: ");
  out.println(MEMPOOL_NUM_MEMBLOCKS - count);
  out.print("free memory: ");
  out.println(MEMPOOL_SIZE - size);
  out.print("addr: ");
  out.println(b->begin + b->size);
}
```

pvrfr, 2021-02-01:
I have added your code to print the state of blocks heap.
I have added some checks to the mempool code to detect any
reentrancy problems.   I will run this until it detects a problem.

In looking at the mempool code I noticed that the resizeBlock
routines could set the block size to zero.   Could this confuse
the allocBlock routine?   I have not studied the code closely
so I may be wrong.

jandrassy, 2021-04-01:
I found the earlier mentioned ENC RAM memory leak cause in EthernetENC and it was with EthernetENC specific code.

pvrfr, 2021-06-13:
I have done a lot more work on the ENC28J60 driver.   I believe that the
problem relates to the lack of a recovery mechanism for a receive buffer
overflow condition.

According the the spec in section 7.2.4 if there is insufficient space
in the receive buffer the RXERIF bit will be set in the EIR register.
Also the BUFER bit will be set in the ESTAT register.   The chip stops receiving data in this overflow state.   This buffer overflow condition and these bits 
will not be cleared without deliberate action by the system.   

There is no code in the driver that does this.   The BUFER bit is not
defined in the header fileenc28j60.h.   The RXERIF bit in the ETR register
is defined in the header file but is not used in the driver code.

A buffer overflow can be a frequent occurrence in some conditions.
The chip supports up to 1000Base-T Ethernet but can only communicate
with the system at up to 20 MHz.   Since there is no on-chip 
processing of packets, all received data must be sent to the system
for handling.   This can easily overload the SPI interface and/or
the system.

The simplest recovery is to call the Enc28J60Network::init routine.
But a more elegant recovery mechanism is needed.

jandrassy, 2021-06-13:
thank you. I check what can be done.

btw: the esp8266 core 3.0.0 is released with support for enc28j60 as interface for LwIP

jandrassy, 2021-06-19:
as I understand the RXERIF is only to indicate the cause of the interrupt if interrupt is used.

where did you find this information: "Also the BUFER bit will be set in the ESTAT register. The chip stops receiving data in this overflow state. This buffer overflow condition and these bits will not be cleared without deliberate action by the system."?

pvrfr, 2021-06-19:
I have read section 12.1.2 RECEIVE ERROR INTERRUPT FLAG (RXERIF) of the manual.
It describes the correct way to recover from a receive buffer overflow.
It says to read the data in the existing buffers, then to advance the buffer
control registers and finally use the BFC command to clear the RXERIF flag.
It also says that a reset can be used.   It is not necessary to clear the
BUFER flag.

jandrassy, 2021-06-20:
RXERIF is about interrupt handling. the library doesn't use enc innterrupt

pvrfr, 2021-06-20:
I understand that RXERIF is in the interrupt status register but it does 
indicate a Rx buffer overflow error just like the BUFER bit in the ESTAT
register.   You can use either one to detect Rx buffer overflow.

The problem is how to handle this error condition.

My system will run for sometimes 30 minutes and other times for a few
months and then will go deaf.   No packets will be received and the
two Rx overflow bits will be set.   I have implemented a recovery
method which just calls the init routine.   That fixes the problem
most of the time but not every time.

I cannot cause the problem to happen, I have to wait for it.

I have a very bad debugging environment.   My OpenSprinkler uses
a ESP8266 module as the processor.  I have tried to connect a debugger
to it without success.   I am left to debug with only the serial port
and a small LCD screen.   

If you have a method of causing the problem and testing the recovery
code that is a big help.

jandrassy, 2021-06-20:
I tested the RX overflow condition. Packets which can't be stored are dropped, but there is no other problem.

pvrfr, 2021-06-20:
When the overflow happens and the packets continue to be received don't 
you have to reset the RXERIF bit?   Does the BUFER bit clear itself?

jandrassy, 2021-06-20:
these bits are flags. they indicate that there was a problem. if the code wants to check the re-ocurence of the problem it has to clear the flag because the enc doesn't clear it. that is all.

pvrfr, 2021-06-20:
Interesting!   The dilemma that I have is why my system goes deaf?   When
this happens the overflow bits are set.   Perhaps setting the bits is
a result of the problem and not the cause.

jandrassy, 2021-06-20:
a basic thing: do you use DHCP for IP address? if yes, do you call Ethernet.maintain to handle DHCP lease renew?

pvrfr, 2021-06-20:
I do not use DHCP, I use a fixed IP address.  And yes, I do call Ethernet.maintain in my polling loop.   During the hang condition the read size returns 0.   That
is the blockSize routine returns 0.

jandrassy, 2021-06-20:
packet counter reaching 255 would block RX. the library does call packet decrement, but maybe there is some bug where it is missed. to debug it you can add following function in Enc28J60Network.cpp  and check the returned value somewhere in your sketch.
uint8_t Enc28J60Network::getRxPacketCount() {
  SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
  uint8_t res = readReg(EPKTCNT);
  SPI.endTransaction();
  return res;
}

pvrfr, 2021-06-20:
We do not use the Arduino environment but I will add that check.   Unfortunately
it can take some time before I get a network going deaf condition.

jandrassy, 2021-06-20:
> it can take some time before I get a network going deaf condition.

the count should be low all the time. you will see if it increases without going back long before it reaches 255.

pvrfr, 2021-06-23:
I have added code that displays EPKTCNT and also records the maximum value that
it hits.   After running for a few days the max is only 1.

Since the buffer in the ENC28J60 is only 8K bytes and is divided between
the transmit and receive buffers the buffer space is severely limited.
The receive buffer is only 2K bytes.   Thus the maximum number of packets
in the buffer is a long way from 255.   Only 25 ICMP packets will fit and
only one full data packet will fit.   Two full data packets will result in
an overflow.

The transmit buffer is 6K which could hold a few data packets.   Perhaps
it would be better to move some of the Tx buffer space to the Rx buffer
space?

I have run for a few days without any errors so I have nothing to report
so far.

jandrassy, 2021-06-23:
but the packets are counted as they are received. if the library doesn't decrement a processed packet in some situation then the counter would increment.

the RX packets for the application are moved into 'TX' memory until the application reads them. that is the reason for large 'TX' memory

pvrfr, 2021-06-27:
I added code to display the value in EPKTCNT and record the max value that
it sees.   I had one hang with the maximum EPKTCNT value at 2.   Most of 
the time it is at 0 and occasionally 1.   However the maximum is currently at 
3 with no hang.

I have put additional code to report some of the other register values.
I am waiting for another hang to see what they will be.

jandrassy, 2021-06-27:
I did more testing. I modify Ethernet library's WebServer example. 

In first test the sketch didn't call the library (the loop returned early) so the RX queue was not handled. I made 255 requests. Then I enabled the rest of the sketch (with Serial input). The counter had 26 RX packets and the library handled them and worked as normal. This test was to check if the EPKTCNT counts packets which are dropped because of full RX memory. It doesn't.

In second test the loop() had Ethernet.maintain() at the beginning so RX packets were handled, but the EthernetServer was stopped so the sketch didn't read the RX packets stored in 'tx' memory managed by MemoryPool. Again I sent 255 http requests. Then I printed the memory pool report. All memory blocks were free. Then I enabled the server handling (with Serial input) and the sketch worked as normal.

pvrfr, 2021-07-01:
I have caught another hang.   This time the current EPKTCNT is 0, the max was 3.

I am beginning to think that I have a memory corruption problem.   Could be
a buffer overflow or possibly some stack corruption.   Without a debugger
like GDB for the ESP8266 it is impossible to look into the machine state
when a hang condition hits.

This system can use the WiFi interface of the ESP8266 for connectivity.
WiFi works perfectly without any hang.   Thus, the problem resides in the
ENC28J60 stack somewhere.

jandrassy, 2021-11-29:
I found a new error. It was hidden in plain sight. Can you try the master version?

matkor, 2022-01-16:
I am fighting similar issue, with Arduino Mega and MySensors.
If fix was included in 2.0.11 version, I am still losing connection, and unable to connect after few hours/days with MySensors protocol over TCP/IP with 2.0.11. :/

jandrassy, 2022-01-16:
> I am fighting similar issue, with Arduino Mega and MySensors. If fix was included in 2.0.11 version, I am still losing connection, and unable to connect after few hours/days with MySensors protocol over TCP/IP with 2.0.11. :/

please, start a new issue and include all info you have.

LeobotInessoft, 2022-02-28:
Not sure if this is directly related to this issue but maybe this will help someone out:
I call " Udp.begin()" multiple times to ensure communication stays active. If I dont do this then the device goes deaf until system reset.
 For example:
   while ((size = Udp.available())>0);
    //finish reading this packet:
    Udp.flush();
    Udp.endPacket();
    Udp.stop();
    Udp.begin(localPort);
    Udp.begin(localPort);
    Udp.begin(localPort);
    Udp.begin(localPort);
    Udp.begin(localPort);

jandrassy, 2022-02-28:
> Not sure if this is directly related to this issue but maybe this will help someone out: I call " Udp.begin()" multiple times to ensure communication stays active. 

for Arduino I recommend the EthernetENC library