UIPEthernet/UIPEthernet #130: Reconect

javierpal, 2021-03-23:
There is a way to reconnect the Ethernet connection if the conection suddenly fails? and if there is one, how I can implement it in my code?.

jandrassy, 2021-03-23:
```
bool ensureConnection() {
  if (!client.connected()) {
    client.stop();
    if (!client.connect(ipAddress, port))
      return false;
  }
  return true;
}
```

javierpal, 2021-03-23:
> ```
> bool ensureConnection() {
>   if (!client.connected()) {
>     client.stop();
>     if (!client.connect(ipAddress, port))
>       return false;
>   }
>   return true;
> }
> ```

i'm using MQTT (client) do you think I can implement that code using that? look here is my code.
`#include <UIPEthernet.h> 
#include <PubSubClient.h>

#define RELAY1 7
#define RELAY2 3
#define RELAY3 5
#define RELAY4 6

/************ Ethernet Conection Information ******************/

byte mac[] = { 0x90, 0xA2, 0xDA, 0x0D, 0x78, 0xEE  };

/************ MQTT Information (CHANGE THESE FOR YOUR SETUP) ******************/
const char* mqtt_server = "";
const char* mqtt_username = "";
const char* mqtt_password = "";
const int mqtt_port = 1883;

/**************************** FOR OTA **************************************************/
#define SENSORNAME "" //change this to whatever you want to call your device

/************* MQTT TOPICS (change these topics as you wish)  **************************/
const char* light_state_topic = "";
const char* light_set_topic = "";

const char* on_cmd = "1";
const char* off_cmd = "0";

#define MQTT_MAX_PACKET_SIZE 512

EthernetClient ethClient;
PubSubClient client;

void setup() {
  Serial.begin(115200);
  
  pinMode(RELAY1,OUTPUT);
  pinMode(RELAY2,OUTPUT);
  pinMode(RELAY3,OUTPUT);
  pinMode(RELAY4,OUTPUT);
  
  pinMode(LED_BUILTIN, OUTPUT);

  setup_ethernet();

  client.setClient(ethClient);
  client.setServer(mqtt_server, mqtt_port);
  client.setCallback(callback);

  
  reconnect();

  Serial.println("Ready");
}

void setup_ethernet() {
  
  Ethernet.begin(mac);
  
  Serial.println("");
  Serial.println("IP address: ");
  Serial.println(Ethernet.localIP());

  digitalWrite(LED_BUILTIN, HIGH);
  delay(100);
  digitalWrite(LED_BUILTIN, LOW);
  delay(100);
  digitalWrite(LED_BUILTIN, HIGH);
}

/********************************** START CALLBACK*****************************************/
void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");

  char message[length + 1];
  for (int i = 0; i < length; i++) {
    message[i] = (char)payload[i];
  }
  message[length] = '\0';
  Serial.println(message);

  int relay = atoi(message);
  switch(relay){
    case 1:
      pulse(RELAY1);
      break;
    case 2:
      pulse(RELAY2);
      break;
    case 3:
      pulse(RELAY3);
      break;
    case 4:
      pulse(RELAY4);
      break;
  }

  client.publish(light_state_topic, message, true);

  digitalWrite(LED_BUILTIN, LOW);
  delay(100);
  digitalWrite(LED_BUILTIN, HIGH);
}

void pulse(int relay) {
  digitalWrite(LED_BUILTIN, LOW);
  delay(100);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(100);
  digitalWrite(relay, HIGH);
  delay(300);
  digitalWrite(relay, LOW);
}

/********************************** START RECONNECT*****************************************/
void reconnect() {
  Serial.print("Attempting MQTT connection...");
  // Attempt to connect
  if (client.connect(SENSORNAME, mqtt_username, mqtt_password)) {
    Serial.println("connected");
    client.subscribe(light_set_topic);
   
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(100);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(100);
    digitalWrite(LED_BUILTIN, HIGH);
  } else {
    Serial.print("failed, rc=");
    Serial.print(client.state());
    Serial.println(" try again in 5 seconds");
    // Wait 5 seconds before retrying
    delay(5000);
  }
}

void loop() {
  if (!client.connected()) {
    reconnect();
  }else{
    client.loop(); 
  }
}`

and sorry for asking but i can't find much docs about the matter.

closed 2021-03-23 by jandrassy