UIPEthernet/UIPEthernet #132: Hangs when connecting on ESP32

bkaraceylan, 2021-05-08:
I am trying to implement an ethernet interface to an existing software. I opted to use ec28j60 with UIPEthernet library. While the provided examples work fine, the same code fails when I implement it in my software. It simply hangs when a connection is attempted. Free heap memory before connection request is 226416. What might be the cause?

platform = espressif32
board = esp32dev
framework = arduino
uipethernet/UIPEthernet@^2.0.9

bkaraceylan, 2021-05-08:
I don't think I do. I initialize the ethernet library after that I wait for a button press to make a connection.

bkaraceylan, 2021-05-08:
I've added Ethernet.maintain() to the loop but the results are the same.

jandrassy, 2021-05-09:
try the EthernetENC library

bkaraceylan, 2021-05-09:
That library doesn't recognize the module for some reason (EthernetNoHardware).

closed 2021-06-19 by jandrassy