UIPEthernet/UIPEthernet #133: STM32F103 ENC28j60

elexal, 2021-05-13:
hi 
when i use this lib get this error
error: 'Enc28J60' was not declared in this scope

**C:\Project\STM32\Ethernet enc28j60\ethernet_enc28j60\ethernet_enc28j60.ino: In function 'void loop()':
ethernet_enc28j60:127:32: error: 'Enc28J60' was not declared in this scope
        uint8_t stateEconRxen = Enc28J60.readReg((uint8_t) NET_ENC28J60_ECON1) & NET_ENC28J60_ECON1_RXEN;**
                                ^
Using library SPI at version 1.0 in folder: C:\Users\xx\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2020.5.26\libraries\SPI 
Using library UIPEthernet at version 2.0.9 in folder: C:\Users\MMAV\Documents\Arduino\libraries\UIPEthernet 
exit status 1
**'Enc28J60' was not declared in this scope**

jandrassy, 2021-05-14:
and why do you have that line in your sketch?

elexal, 2021-05-14:
i deleted that code
but in example : ethernet advance web server i get error 

In file included from C:\Users\MMAV\AppData\Local\Temp\arduino_modified_sketch_905279\STM32-Ethernet_AdvancedWebServer.ino:21:0:
C:\Users\MMAV\Documents\Arduino\libraries\EthernetWebServer_STM32\src/EthernetWebServer_STM32.h:36:3: error: #error This code is designed to run on STM32F/L/H/G/WB/MP1 platform! Please check your Tools->Board setting.
  #error This code is designed to run on STM32F/L/H/G/WB/MP1 platform! Please check your Tools->Board setting.
   ^
In file included from C:\Users\MMAV\AppData\Local\Temp\arduino_modified_sketch_905279\STM32-Ethernet_AdvancedWebServer.ino:21:0:
C:\Users\MMAV\Documents\Arduino\libraries\EthernetWebServer_STM32\src/EthernetWebServer_STM32.h:74:22: fatal error: pgmspace.h: No such file or directory
 #include <pgmspace.h>
                      ^
compilation terminated.
Using library EthernetLarge-master at version 2.0.0 in folder: C:\Users\MMAV\Documents\Arduino\libraries\EthernetLarge-master 
Using library WebSockets2_Generic-master at version 1.2.4 in folder: C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master 
Using library EthernetWebServer_STM32 at version 1.2.0 in folder: C:\Users\MMAV\Documents\Arduino\libraries\EthernetWebServer_STM32 
Using library Functional-Vlpp at version 1.0.2 in folder: C:\Users\MMAV\Documents\Arduino\libraries\Functional-Vlpp 
exit status 1
Error compiling for board Generic STM32F103C series.

elexal, 2021-05-14:
"C:\\Users\\MMAV\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10813 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ "-IC:\\Users\\MMAV\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2020.5.26\\system/libmaple" "-IC:\\Users\\MMAV\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2020.5.26\\system/libmaple/include" "-IC:\\Users\\MMAV\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2020.5.26\\system/libmaple/stm32f1/include" "-IC:\\Users\\MMAV\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2020.5.26\\system/libmaple/usb/stm32f1" "-IC:\\Users\\MMAV\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2020.5.26\\system/libmaple/usb/usb_lib" "-IC:\\Users\\MMAV\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2020.5.26\\cores\\maple" "-IC:\\Users\\MMAV\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2020.5.26\\variants\\generic_stm32f103c" "-IC:\\Users\\MMAV\\Documents\\Arduino\\libraries\\EthernetLarge-master\\src" "-IC:\\Users\\MMAV\\Documents\\Arduino\\libraries\\WebSockets2_Generic-master\\src" "-IC:\\Users\\MMAV\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F1\\2020.5.26\\libraries\\SPI\\src" "C:\\Users\\MMAV\\AppData\\Local\\Temp\\arduino_build_54859\\sketch\\STM32-Ethernet-Minimal-Client.ino.cpp" -o "C:\\Users\\MMAV\\AppData\\Local\\Temp\\arduino_build_54859\\sketch\\STM32-Ethernet-Minimal-Client.ino.cpp.o"
In file included from C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/WebSockets2_Generic.h:42:0,
                 from C:\Users\MMAV\AppData\Local\Temp\arduino_modified_sketch_133693\STM32-Ethernet-Minimal-Client.ino:41:
C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/Tiny_Websockets_Generic/server.hpp:77:51: error: expected ')' before '*' token
       WebsocketsServer(network2_generic::TcpServer* server = new WSDefaultTcpServer);
                                                   ^
C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/Tiny_Websockets_Generic/server.hpp:93:7: error: 'TcpServer' in namespace 'websockets2_generic::network2_generic' does not name a type
       network2_generic::TcpServer* _server;
       ^
In file included from C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/WebSockets2_Generic.h:45:0,
                 from C:\Users\MMAV\AppData\Local\Temp\arduino_modified_sketch_133693\STM32-Ethernet-Minimal-Client.ino:41:
C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/WebSockets2_Generic_Client.hpp: In constructor 'websockets2_generic::WebsocketsClient::WebsocketsClient()':
C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/WebSockets2_Generic_Client.hpp:51:76: error: 'WSDefaultTcpClient' was not declared in this scope
   WebsocketsClient::WebsocketsClient() : WebsocketsClient(std::make_shared<WSDefaultTcpClient>())
                                                                            ^
C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/WebSockets2_Generic_Client.hpp:51:96: error: no matching function for call to 'make_shared()'
   WebsocketsClient::WebsocketsClient() : WebsocketsClient(std::make_shared<WSDefaultTcpClient>())
                                                                                                ^
C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/WebSockets2_Generic_Client.hpp:51:96: note: candidate is:
In file included from c:\users\mmav\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\memory:82:0,
                 from C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/Tiny_Websockets_Generic/internals/websockets_endpoint.hpp:39,
                 from C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/Tiny_Websockets_Generic/client.hpp:40,
                 from C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/WebSockets2_Generic.h:41,
                 from C:\Users\MMAV\AppData\Local\Temp\arduino_modified_sketch_133693\STM32-Ethernet-Minimal-Client.ino:41:
c:\users\mmav\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\bits\shared_ptr.h:610:5: note: template<class _Tp, class ... _Args> std::shared_ptr<_Tp1> std::make_shared(_Args&& ...)
     make_shared(_Args&&... __args)
     ^
c:\users\mmav\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\bits\shared_ptr.h:610:5: note:   template argument deduction/substitution failed:
In file included from C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master\src/WebSockets2_Generic.h:45:0,
[... cut ...]
     this->_server->close();
           ^
Using library EthernetLarge-master at version 2.0.0 in folder: C:\Users\MMAV\Documents\Arduino\libraries\EthernetLarge-master 
Using library WebSockets2_Generic-master at version 1.2.4 in folder: C:\Users\MMAV\Documents\Arduino\libraries\WebSockets2_Generic-master 
Using library SPI at version 1.0 in folder: C:\Users\MMAV\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2020.5.26\libraries\SPI 
exit status 1
Error compiling for board Generic STM32F103C series.

jandrassy, 2021-05-14:
did you set it to use UIPEthernet?

closed 2021-06-19 by jandrassy