UIPEthernet/UIPEthernet #134: PlatformIO isn't pulling the 2.0.9 version

melyux, 2021-05-21:
The PIO library is stuck on 2.0.8. The only thing I can think of is that the new `library.json` with the multiple authors file should have commas between the curly bracket sections, like in [this example(https://docs.platformio.org/en/latest/librarymanager/config.html#authors).

Right now, there are no commas.

https://github.com/UIPEthernet/UIPEthernet/commit/b9203c78b0f4e33651fb33bfe3e3a9a844888a97#diff-64cc7cdde6789dc1c1c4a6f406cd1ff6a6a00027097788e2d52fc27b0d75502eR12-R29

Could be another problem, I don't know how strict PIO crawler is about the formatting of the JSON, but that's my first guess.

jandrassy, 2021-05-21:
yes there should be commas

melyux, 2021-06-04:
It's still not pulling the newer versions after 2.0.8. Looks like it wasn't the comma issue after all.

jandrassy, 2021-06-04:
> It's still not pulling the newer versions after 2.0.8. Looks like it wasn't the comma issue after all.

maybe ask at platformio

2bndy5, 2021-07-11:
@jandrassy @cassy73 v2.0.8 is now showing up, but nothing after that.

I'm writing to tell you I just had a similar experience with the RF24 repo: a missing comma in json required manual re-publishing the lib to PIO registry. However, PIO's new "trusted" lib registry is moving away from crawling existing ("legacy") libs for new releases. You may have to automate publishing this lib yourselves from now on.

closed 2022-01-16 by jandrassy

melyux, 2022-01-16:
Any reason for closure?

jandrassy, 2022-01-17:
is it no fixed? Platformio's Ivan Kravets fixed libarary.json
https://github.com/UIPEthernet/UIPEthernet/pull/145

2bndy5, 2022-01-17:
Well, v2.0.9 and v2.0.10 are not available on PIO, but v2.0.11 is.