UIPEthernet/UIPEthernet #139: Unstable behavior

sbruimen, 2021-07-27:
Hello, dear experts.

I have a sketch and hardware (arduino nano + ENC28J60 nano shield). About 10 clients system installations. 
Using UIPethernet library. It works fine in a 9 client networks. And in my home test network the system works fine without failures for years uptime.

But recently we met a new client, in their network system is unstable and randomly lost connectivity from 2 to 24 hours.

I have activated debug mode level 1 as the nano memory allows.

Can you please inspect the last minutes of system life log, and can you explain me, why the problems are occurring?
[log.txt](https://github.com/UIPEthernet/UIPEthernet/files/6886256/log.txt)
My competence is not enough to understand the UIPethernet log debug information.

What i noticed: in the log file, starting from 05:30:21.553 appears "Packet count ## -> failed " , then the packet count value is decremented step by step -1, from 99 to 1, and then the system lost connection (function client.connect(IP,port) return 0 permanently)

What could be the reason in client network or in system or UIP library for this behavior?

PS: in the log "FM: 498" is a Free Memory. It is same and stable in other system installations.

jandrassy, 2021-07-28:
the library is not called frequently enough here to handle the network traffic. the log shows library calls less than every second. the count of unprocessed packets in receive memory of enc  then increments. the enc writes them and the library doesn't read them fast enough. these could be some broadcast packets in the network and the library needs to just skip them.
if the receive buffer is full the enc stops receiving packets which cause the following problems.
if there is no other frequent library call, call Ethernet.maintain() in loop() and make sure the loop loops fast enough so the library is called at least every 200 milliseconds.
other option is to filter broadcasts in enc so the broadcast packets are not put in the RX buffer. the some services relaying on receiving broadcast like DHCP will not work.
my EthernetENC library has broadcast filtered in ENC and DHCP modified to use unicast response. but this doesn't work with some home routers.

sbruimen, 2021-07-29:
Thank you very much for your answer! 

Before writing a query here, I used a Wireshark tool to analyse traffic. There are several "smart" printers and other devices connected on the client's local network that generate a lot of broadcast traffic. Probably at some moments this exceeds the threshold value, that my SYSTEM can process.

The problem became clearer. I went to rewrite the sketch.

closed 2021-07-29 by sbruimen