UIPEthernet/UIPEthernet #147: not getting ip adress with esp32

Mnasri97, 2022-01-27:
Hi,
i'm using enc28j60 with the esp32

My connections are as follows:
SO--> MISO pin 19
 SI--> MOSI pin 23 
CS--> CS pin 5
SCK-->CLK pin18

I uploaded this piece of code to the esp32:

```
#include <UIPEthernet.h>
#include <utility/logging.h>
#include <SPI.h>
//#define cs 5
EthernetClient ethClient;
uint8_t macaddress[6] = {0x1,0x2,0x3,0x4,0x5,0x6};
void setup() {
    Serial.begin(115200);
    // Ethernet.init(cs);
    Ethernet.begin(macaddress); 
    delay(5000);
    Serial.print("Local IP : ");
    Serial.println(Ethernet.localIP());
    Serial.print("Subnet Mask : ");
    Serial.println(Ethernet.subnetMask());
    Serial.print("Gateway IP : ");
    Serial.println(Ethernet.gatewayIP());
    Serial.print("DNS Server : ");
    Serial.println(Ethernet.dnsServerIP());
   Serial.println("Ethernet Successfully Initialized");
}
void loop() {
}
``` 

output data is:

```
Local IP : 0.0.0.0
Subnet Mask : 0.0.0.0
Gateway IP : 0.0.0.0
DNS Server : 0.0.0.0
Ethernet Successfully Initialized
```
what am I doing wrong ?

jandrassy, 2022-01-27:
```
  Serial.println("Initialize Ethernet with DHCP:");
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
      }
    }
    if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }
    // try to configure using IP address instead of DHCP:
    Ethernet.begin(mac, ip, myDns);
  } else {
    Serial.print("  DHCP assigned IP ");
    Serial.println(Ethernet.localIP());
  }
 ```

Mnasri97, 2022-01-27:
> ```
>   Serial.println("Initialize Ethernet with DHCP:");
>   if (Ethernet.begin(mac) == 0) {
>     Serial.println("Failed to configure Ethernet using DHCP");
>     // Check for Ethernet hardware [...]

can you please explain to me what i should do  ?

jandrassy, 2022-01-27:
it should diagnose your problem

Mnasri97, 2022-01-27:
it tells me 
`Ethernet shield was not found.  Sorry, can't run without hardware. :(`

jandrassy, 2022-01-27:
try ` Ethernet.init(5);`