UIPEthernet/UIPEthernet #148: 2.0.11 release break my arduino nano MQTT connection

thalesmaoa, 2022-02-11:
I have some Arduino Nano mounted over a ENC28L60 shield. The goal is to publish MQTT over the network.
Recently I decided to make some changes and upgraded the UIPEthernet lib, however MQTT doesn't connect anymore.

```
$ diff UIPEthernet/ ~/Downloads/tmp/src/UIPEthernet/
Subdiretórios idênticos: UIPEthernet/examples e /home/thales/Downloads/tmp/src/UIPEthernet/examples
Somente em /home/thales/Downloads/tmp/src/UIPEthernet/: .git
Subdiretórios idênticos: UIPEthernet/hardware e /home/thales/Downloads/tmp/src/UIPEthernet/hardware
diff UIPEthernet/library.json /home/thales/Downloads/tmp/src/UIPEthernet/library.json
3c3
<   "keywords": "ENC28j60, Ethernet, UIP, TCP/IP, Network",
---
>   "keywords": "ENC28j60, Ethernet, UIP, TCP, IP, Network",
17c17
<     }
---
>     },
23c23
<     }
---
>     },
31c31
<   "version": "2.0.9",
---
>   "version": "2.0.11",
33a34
>   "headers": "UIPEthernet.h",
diff UIPEthernet/library.properties /home/thales/Downloads/tmp/src/UIPEthernet/library.properties
8c8
< version=2.0.9
---
> version=2.0.11
Subdiretórios idênticos: UIPEthernet/mbed e /home/thales/Downloads/tmp/src/UIPEthernet/mbed
diff UIPEthernet/README.md /home/thales/Downloads/tmp/src/UIPEthernet/README.md
5a6,7
> For new projects with enc28j60 EthernetENC library is recommended.
> 
88a91,101
> 
> - make hardwareStatus() and linkStatus() work without Erhernet.begin(mac)
> - UIPServer.end() - stop() all clients
> - fixed library.json for platformio
> - New release:2.0.10
> 
> - UIPUdp.cpp beginPacket fix IPAddress check for esp8266 core 3.x.x
> - platformio library.json fixes by Ivan Kravets
> - UIPClient._allocateData - fix data clear
> - UIPClient - fix NULL data field checks
> - New release:2.0.11
Subdiretórios idênticos: UIPEthernet/tests e /home/thales/Downloads/tmp/src/UIPEthernet/tests
diff UIPEthernet/UIPClient.cpp /home/thales/Downloads/tmp/src/UIPEthernet/UIPClient.cpp
293c293
<   if (*this)
---
>   if (data)
318c318
<   if (*this)
---
>   if (data)
373c373
<   if (*this)
---
>   if (data)
391c391
<   if (*this)
---
>   if (data)
585a586
>           memset(data, 0, sizeof(uip_userdata_t));
588d588
<           memset(&data->packets_in[0],0,sizeof(uip_userdata_t)-sizeof(data->state));
diff UIPEthernet/UIPUdp.cpp /home/thales/Downloads/tmp/src/UIPEthernet/UIPUdp.cpp
92c92
<   if (ip && port)
---
>   if ((ip[0] || ip[1] || ip[2] || ip[3]) && port)
Subdiretórios idênticos: UIPEthernet/utility e /home/thales/Downloads/tmp/src/UIPEthernet/utility

```

I think that is related to memset.

thalesmaoa, 2022-02-11:
I confirm

```
memset(data, 0, sizeof(uip_userdata_t));
```
doesn't work.

The code bellow works.
```
memset(&data->packets_in[0],0,sizeof(uip_userdata_t)-sizeof(data->state));
```

jandrassy, 2022-02-12:
You must have some bug in your sketch, which a bug in library was covering.
The memset in _allocateData had to be fixed. After conn_index was added to uip_userdata_t in v 2.0.6, the memset was writing one byte after the struct. it caused clearing conn_index of other connection.

thalesmaoa, 2022-02-13:
Ok, let me try to help. 
My code is just following other examples found online, as example:
https://github.com/zuhryfayesz/Zuhry-MQTT-IoT

If there is a bug, it can be related to PubSubClient as well.
It does connect to the internet, however it gets timeout to mqtt server.

Here is my code:

```
//#define UIP_CONF_UDP             1
#include "src/UIPEthernet/UIPEthernet.h" 
#include "src/pubsubclient/PubSubClient.h"

//#define DEBUG
#define ARDUINO 1
#define USINA 1

#if ARDUINO == 1
  #define CLIENT_ID "ard01"
  #define IP_ADDR 1, 61
#else if ARDUINO == 2
  #define CLIENT_ID "ard02"
  #define IP_ADDR 1, 62
#endif

#define PUBLISH_DELAY   5000

#if USINA
  #define IP_SERVER 1, 209
#else
  #define IP_ADDR 200, 84
  #define IP_SERVER 200, 254
#endif

#define MQTT_USER "xxxx"
#define MQTT_PASS "xxxx"

int  upTime = 0;
byte mac[] = { 0xDE, 0xED, 0xBA, 0xFE, 0xFE, 0xED };

IPAddress ip(192,168,IP_ADDR);
IPAddress server(192, 168, IP_SERVER);

EthernetClient ethClient;
PubSubClient mqttClient;

void setup() {
  // setup serial communication
  #ifdef DEBUG
    Serial.begin(9600);
  #endif
    
  Ethernet.begin(mac, ip);
  delay(1500);
  
  // setup mqtt client
  mqttClient.setClient(ethClient);
  mqttClient.setServer(server, 1883);
  #ifdef DEBUG
    Serial.println(F("MQTT client configured"));
    Serial.print(CLIENT_ID);
    Serial.print(" - ");
    Serial.println(ip);
    Serial.print("Local: ");
    if (USINA){
      Serial.print("Usina - ");
    }else{
      Serial.print("Casa - ");
    }
    Serial.println(server);
  #endif
  
  delay(1500);

  #ifdef DEBUG
    Serial.println(F("Ready to send data"));
  #endif
}

void loop() {
  if (!mqttClient.connected()) {    reconnect();  }
  mqttClient.loop();
  }
}

void reconnect() {
  uint8_t conec = 0;
  // Loop until we're reconnected
  while (!mqttClient.connected()) {
    #ifdef DEBUG
      Serial.print("Attempting MQTT connection...");
    #endif
    // Attempt to connect
    if (mqttClient.connect(CLIENT_ID, MQTT_USER, MQTT_PASS)) {
      #ifdef DEBUG
        Serial.println("connected");
      #endif
    } else {
      #ifdef DEBUG
        Serial.print("failed, rc=");
        Serial.print(mqttClient.state());
        Serial.println(" try again in 5 seconds");
        // Wait 5 seconds before retrying
      #endif
      delay(5000);
    }
  }
}
```

jandrassy, 2022-02-13:
can you please try the EthernetENC library?

thalesmaoa, 2022-02-13:
I confirm that using EthernetENC does work. This means that I no longer need  UIPEthernet, correct?

jandrassy, 2022-02-13:
> I confirm that using EthernetENC does work. This means that I no longer need UIPEthernet, correct?

yes. 
and I must find the next bug in UIPEthernet