UIPEthernet/UIPEthernet #152: SPI class conflict with "Arduino Samd board 1.8.13"

laszlokovacs25, 2022-03-23:
Hi,

I developed a project with samd xplained pro board, and the code has been compiled successfully until the "samd board" version was 1.8.11 but with the version 1.8.13 your library has conflict with SPI class.

"
C:\Users\laszl\OneDrive\Documents\Arduino\libraries\UIPEthernet\utility\Enc28J60Network.cpp:44:24: error: conflicting declaration 'arduino::SPIClass SPI'
        extern SPIClass SPI;
                        ^~~
In file included from C:\Users\laszl\OneDrive\Documents\Arduino\libraries\UIPEthernet\utility\Enc28J60Network.h:211:0,
                 from C:\Users\laszl\OneDrive\Documents\Arduino\libraries\UIPEthernet\utility\Enc28J60Network.cpp:25:
C:\Users\laszl\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13\libraries\SPI/SPI.h:93:23: note: previous declaration as 'SPIClassSAMD SPI'
   extern SPIClassSAMD SPI;
                       ^~~ 
"

can you please fix this UIPEthernet library to be compatible with the new version samd board.

Thank you

[image]

jandrassy, 2022-03-23:
please use the EthernetENC library

jandrassy, 2022-03-23:
can you switch to EthernetENC library?
https://github.com/jandrassy/EthernetENC/wiki