UIPEthernet/UIPEthernet #153: Can't receive UDP Broadcasts

leelx78, 2022-05-01:
Hi I'm trying to receive UDP broadcasts from 10.7.255.255:6000 using ENC28J60 on ESP32S2 but I get nothing.

On normal UDP Unicast I have no issues, e.g. using

`udp.begin(UDP_PORT);
`

but for broadcasts and multicasts I used to initialise it this way on NativeEthernet and Teensy 
`udp.beginMulticast(IPAddress(10,7,255,255), UDP_PORT);
`

I looked at Dhcp.cpp and from there the device SHOULD be able to receive broadcasts, otherwise it would not get an IP, correct? But There is just seems to work using the begin method

`if (_dhcpUdpSocket.begin(DHCP_CLIENT_PORT) == 0)
    {
      // Couldn't get a socket
      return 0;
    }
`
Can anybody please help me to understand how to receive broadcasts here?

Thanks a lot!

JAndrassy, 2022-05-02:
UIPEthernet doesn't support beginMulticast, but multicast is something else than broadcast. to broadcast you only have to use broadcast address.

leelx78, 2022-05-02:
Thanks @JAndrassy . 

So are you saying that if I initialise this way:

```
EthernetUDP udp;

Ethernet.begin(mac, IPAddress(10,0,0,3), IPAddress(10,0,0,1), IPAddress(10,0,0,2), IPAddress(255,248,0,0));

udp.begin(UDP_PORT);
```

and then I read in loop() this way:
```
bool readUDP() {
  int packetSize = udp.parsePacket();
  if (packetSize) {
    Serial.print("Received packet of size ");
    Serial.println(packetSize);
    Serial.print("From ");
    IPAddress remote = udp.remoteIP();
    for (int i=0; i < 4; i++) {
      Serial.print(remote[i], DEC);
      if (i < 3) {
        Serial.print(".");
      }
    }
    Serial.print(", port ");
    Serial.println(udp.remotePort());
    udp.read(packetBuffer, packetSize);
    Serial.println("Contents:");
    Serial.println(packetBuffer);

  }else{
    //Serial.println("no data");
  }
  
  delay(10);
  return false;
}
```

it should work? I have no problem in getting UDP packets addressed to 10.0.0.3 but anything on 10.7.255.255 (which is the broadcast address for the 255.248.0.0 net) is completely ignored.

What am I missing?

Thanks again!

leelx78, 2022-05-02:
some more info. Looks like I can read data from 10.7.255.255 now but there is a weird bug though, I.e. I can only read data from ONE source. If I have 2 devices broadcasting on 10.7.255.255 I only read data from the FIRST one.

Any idea why this happens?

JAndrassy, 2022-05-02:
it is not bug, but a feature. So I can't change it. My EthernetENC library has better UDP receiving, but has broadcast filtered out in ENC.

leelx78, 2022-05-02:
Hi @JAndrassy hang on, I am actually able to receive broadcasts now so something must be working.  

Unfortunately 
- when I listen to some encrypted long packets (~80 bytes payload) I get always only 2 of them, then nothing anymore.
- If I listen to other shorter non-encrypted broadcasts it goes on forever as long as the source is not restarted. (I have a Python script to test this)

Do you have any idea of what could cause UDP to crash or stop receiving packets? Is there a way to reset the udp object?

I think this also affects unicasts so it might be some unhandled exception in the UDP class?

leelx78, 2022-05-02:
Also, if I try to use EthernetENC:

- does it work for ESP32?
- can I un-filter broadcasts by editing the code? Some hint on where to edit would be great!

JAndrassy, 2022-05-02:
> Now I'm looking in the code if I can un-filter broadcasts. Any suggestion on where to start?

https://github.com/JAndrassy/EthernetENC/blob/a0393889ea90ea17e7ab8c33c759e1d6f16748e1/src/utility/Enc28J60Network.cpp#L98

leelx78, 2022-05-02:
Thanks @JAndrassy

Ok so it's a matter of changing this
```
  // do bank 1 stuff, packet filter:
  // For broadcast packets we allow only ARP packtets
  // All other packets should be unicast only for our mac (MAADR)
  //
  // The pattern to match on is therefore
  // Type     ETH.DST
  // ARP      BROADCAST
  // 06 08 -- ff ff ff ff ff ff -> ip checksum for theses bytes=f7f9
  // in binary these poitions are:11 0000 0011 1111
  // This is hex 303F->EPMM0=0x3f,EPMM1=0x30
  writeReg(ERXFCON, ERXFCON_UCEN|ERXFCON_CRCEN|ERXFCON_PMEN);
  writeRegPair(EPMM0, 0x303f);
  writeRegPair(EPMCSL, 0xf7f9);
```
 
I did some trial-and-error in commenting out the 3 bottom lines hoping for a lucky strike but without success. Can you please advise the correct command for this? It's not laziness on my end, I just have no experience in coding the ENC interface so you definitely have a clearer view on this thanks!

JAndrassy, 2022-05-02:
https://github.com/UIPEthernet/UIPEthernet/blob/e7a117b0eb2f4535bc1d276a69fea02d3912d20d/utility/Enc28J60Network.cpp#L276

leelx78, 2022-05-02:
thanks. I changed the file and recompiled with no apparent difference but that allows me to focus elsewhere. For instance, for broadcasts on a subnet 255.248.0.0 I get broadcast address 10.7.255.255. GW is 10.0.0.2 and DNS 10.0.0.1. So should I start Ethernet this way

```
Ethernet.begin(mac, IPAddress(10,0,0,3), IPAddress(10,0,0,1), IPAddress(10,0,0,2), IPAddress(255,248,0,0));
```

or using DHCP
```
Ethernet.begin(mac);
 ```

None worked so far but maybe one is a more correct approach for this.

As for reading out, all I'm doing is 
```
    udp.begin(UDP_PORT);
```
in the setup() function, i.e. done once and then periodically calling readUDP() in the loop function. Is there anything fundamentally wrong so far?

JAndrassy, 2022-05-02:
it is is always better to use DHCP with Ethernet.begin(mac); and then print the IPAddress.

for UDP the Ethernet library's UDPSendReceiveString example shows the usage. it should work for broadcasted UDP too. 

(btw: DHCP can use unicast (to MAC address) and the EthernetENC library sets the DHCP unicast flag to get the response as unicast)

leelx78, 2022-05-03:
looking at UDPSendReceiveString.ino I'm doing pretty much the same thing but anything sent to 10.7.255.255 is not received.

do you have a chance to test it with a simple Python script at your end? If I send packets addressed to there device IP it works perfectly but everything sent on the broadcast address (in my case 10.7.255.255) is ignored. Note that if you try on your network you might need to change the BC address

```
import socket, time
DEST="10.7.255.255" #if you put the device IP here it will work perfectly
PORT=5000

s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
s.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, 1)
s.settimeout(0.1);

for i in range(1000000):
	print "Sending data to %s:%d"%(DEST, PORT)
	try:
		s.sendto("CPU SEND:%s\n"%time.ctime(), (DEST, PORT))
	except:
		print "ERROR"
		pass
	time.sleep(1)
s.close()

```

JAndrassy, 2022-05-03:
I think in local network the broadcast address is 255.255.255.255. 10.7.255.255 would be necessary if it should be routed to other network

leelx78, 2022-05-03:
right, some progress here.

First, I needed to go back to **UIPEthernet** and eventually I managed to get broadcasts with that library on 10.7.255.255 as long as I call this after every packet is received:

```
      udp.stop();
      udp.begin(UDP_PORT);
```
      
If I don't reset UDP it will only listed to a single server and ignore all other devices broadcasting on the network. With this it listens to all but it looks VERY wasteful. Is the fact that the UDP class gets "locked" to the first sender it receives intentional or a bug?

As for broadcasting, I ended up having to do define a separate UDP object for outwards traffic 

```
EthernetUDP udp; //input
EthernetUDP udp_out; //output
```

and do this to avoid lock-ups:

```
  udp.stop();
  delay(1);
  Serial.println("Broadcasting data...");

  udp_out.beginPacket(BROADCAST_IP, UDP_PORT);
  udp_out.write(ReplyBuffer);
  udp_out.endPacket();

  udp.begin(UDP_PORT);
```

Again, it looks pretty ugly but it's working somehow. Any comments?

closed 2022-05-04 by JAndrassy

sdrshnptl, 2023-04-02:
> right, some progress here.
> 
> First, I needed to go back to **UIPEthernet** and eventually I managed to get broadcasts with that library on 10.7.255.255 as long as I call this after every packet is received:
> 
> ```
>       udp.stop();
>       udp.begin(UDP_PORT);
> ```
> 
> If I don't reset UDP it will only listen to a single server and ignore all other devices broadcasting on the network. With this it listens to all but it looks VERY wasteful. Is the fact that the UDP class gets "locked" to the first sender it receives intentional or a bug?
> 
> As for broadcasting, I ended up having to do define a separate UDP object for outwards traffic
> 
> ```
> EthernetUDP udp; //input
> EthernetUDP udp_out; //output
> ```
> 
> and do this to avoid lock-ups:
> 
> ```
>   udp.stop();
>   delay(1);
>   Serial.println("Broadcasting data...");
> 
>   udp_out.beginPacket(BROADCAST_IP, UDP_PORT);
>   udp_out.write(ReplyBuffer);
>   udp_out.endPacket();
> 
>   udp.begin(UDP_PORT);
> ```
> 
> Again, it looks pretty ugly but it's working somehow. Any comments?

Hi! @leelx78 ,
Can you share complete example of UDP broadcast?
is there library level changes you've made?