UIPEthernet/UIPEthernet #154: Adding a new MAC-PHY chip

vChavezB, 2022-05-02:
I am currently working in a research group where we want to develop an Arduino framework for the latest ethernet PHY, [10BASE-T1L](https://www.analog.com/media/en/technical-documentation/tech-articles/enabling-seamless-ethernet-to-field-with-10base-t1l-connectivity.pdf).

We plan to integrate lwIP but we are also interested in using small MCUs, such as with your integration of UIP. Since you have already implemented it, we want to check if we can modify the class `Enc28J60Network ` and use our Ethernet Mac-PHY driver instead.

For this reason, I want to ask how easy would it be to change the class `Enc28J60Network ` implementation and replace the public methods. I did not find documentation for this class, so after checking some of the references and parts where these methods are used I have the following notes. Correct me if I am wrong:

-   `receivePacket`
    - Returns `NOBLOCK `if no ethernet packet has been received.
    - Returns `UIP_RECEIVEBUFFERHANDLE` if packet(s) have been received, `receivePkt.size` must be updated with the total length of the packet
-   `readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len)`
    - Used only after checking that  `Enc28J60Network ::receivePacket()` has returned `UIP_RECEIVEBUFFERHANDLE`.
    - `handle` argument just returns the value received from  `Enc28J60Network ::receivePacket()`.
    - `position` argument is the position in the receive buffer where to start reading.
    - `buffer` Buffer where to write the received package.
    - `len` Total bytes to write to `buffer`
-   freePacket(void)
    -  Deletes a packet from the RX buffer queue of the MAC-PHY.
-   memaddress blockSize(memhandle handle)
    - Returns 0  if no packet has been received. This happens when handle == NOBLOCK or erevid == 0 which means there is no communication with the MAC-PHY chip.
    - Returns `receivePkt.size` if there is a packet in the buffer.
    - Else returns  blocks[handle].size. 
-   sendPacket(memhandle handle)
    - Transmits a packet with the Ethernet MAC-PHY.
    - Uses the latest packet in the buffer wrote with `Enc28J60Network ::writePacket`
    - Returns `true` if transmission sucessfull
-   `writePacket`(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len)
    - Writes a packet to the TX buffer of the MAC-PHY.
    - `handle` refers to the index from `struct memblock blocks[]`
    - `position` is the offset where to write in the block
    - `buffer` Data to use for writing into the block
    - `len` Total bytes to write to buffer
-   copyPacket(memhandle dest, memaddress dest_pos, memhandle src, memaddress src_pos, uint16_t len)
    -  Copies a packet to the Write Buffer memory of the MAC-PHY
    - If `src_pkt` is UIP_RECEIVEBUFFERHANDLE , the source is `receivePkt` otherwise `blocks[src_pkt]`
-   chksum(uint16_t sum, memhandle handle, memaddress pos, uint16_t len);
    - Creates a checksum from the Read buffer memory of the MAC-PHY

JAndrassy, 2022-05-02:
this library uses very specific features of the Atmel ENC28j60. While the ENC28j60 can work at MAC level, this library uses the additional support level of the ENC2860.

closed 2022-05-02 by JAndrassy

JAndrassy, 2022-05-02:
esp8266 Arduino supports Ethernet MAC level network adapters for network interfaces with LwIP. Currently W5500 and ENC28j60 are supported.
I have a dream to create a multi-platform Arduino networking library based on LwIP and supporting MAC level WiFi and Ethernet adapters. It would adopt the interface implementations used by the esp8266 Arduino.

vChavezB, 2022-05-02:
> this library uses very specific features of the Atmel ENC28j60. While the ENC28j60 can work at MAC level, this library uses the additional support level of the ENC2860.

Could you mention what are the additional support features that are used? 

From what I have checked so far, it has a buffer memory and DMA access to the buffer. The buffer is built with the public methods and then sent/recieveed to the ENC28j60.

vChavezB, 2022-05-02:
>create a multi-platform Arduino networking library based on LwIP and supporting MAC level WiFi and Ethernet adapters. It would adopt the interface implementations used by the esp8266 Arduino.

I have created a [LwIP port ](https://github.com/vChavezB/lwip-Arduino) that is compatible with Arduino. This purpose of this port is to expose the LwIP API, so use cases like the one you mention can be implemented.

JAndrassy, 2022-05-02:
>  From what I have checked so far, it has a buffer memory and DMA access to the buffer. The buffer is built with the public methods and then sent/recieveed to the ENC28j60.

The 8kB RAM of the enc28j60 is used for received and composed  packets.