UIPEthernet/UIPEthernet #155: Can't get any data from HTTP connection

leelx78, 2022-05-06:
Hi I have an issue het I can connect to a server, send an HTTP request and always get nothing back:

```
bool httpRequest() {
  if (client.connect(IPAddress(10, 0, 0, 1), 8080)) {
    Serial.print("Sending HTTP Request...");
    delay(10);
    client.println("GET / HTTP/1.1");
    client.println("Host: www.example.com");
    client.println("User-Agent: arduino-ethernet");
    client.println("Connection: close");
    client.println();

    Serial.println("...OK - Response:");
    long tmax = millis() + 2000;
    while (millis() < tmax) {
      delay(1);
      if (client.available()) {
        char c = client.read();
        Serial.print(c);
      }
    }
  } else {
    Serial.println("httpRequest >> FAILED TO CONNECT");
    client.stop();
    return false;
  }
  return true;
}
``` 

and when I get is always this:

```
Sending HTTP Request......OK - Response:
Sending HTTP Request......OK - Response:
Sending HTTP Request......OK - Response:
Sending HTTP Request......OK - Response:
Sending HTTP Request......OK - Response:
Sending HTTP Request......OK - Response:
Sending HTTP Request......OK - Response:
Sending HTTP Request......OK - Response:
```

I tried with EthernetENC and this works fine but as per previous communication I cannot use EthernetENC because it's not working with UDP broadcasts. Can anybody help me either in getting UIPEthernet work for this or to unblock the UDP broadcasts in EthernetENC

Thanks!

leelx78, 2022-05-07:
Also tried using a TCP connection on a server in Python and I confirm I get absolutely nothing back with UIPEthernet.

Changing to EthernetENC using the same code works perfectly on TCP but unfortunately breaks the code on UDP broadcasts

Looking forward to a solution for this because I'm completely stuck :|

JAndrassy, 2022-05-07:
what version of UIPEthernet you use?

leelx78, 2022-05-07:
Installed from Arduino Library manager:
- UIPEthernet v 2.0.11
- EthernetENC v2.0.2

Getting nothing back from TCP seems too strange to be true, it would be a huge bug which would never pass testing. So I must be doing something wrong but honestly I don't know what. The same code works perfectly with EthernetENC.

Can you try to get the content of a web page (e.g. the raw response) as a test using UIPEthernet? Would be curious to see if you see what I see. I'm on ESP32-S2 if that matters.

Thanks!

JAndrassy, 2022-05-07:
please try the GitHub master version of UIPEthernet. there are some fixes

leelx78, 2022-05-07:
Thanks so much @JAndrassy ! The GitHub works perfectly for both!

closed 2022-05-07 by leelx78

JAndrassy, 2022-05-08:
2.0.12 should be available in Library Manager in a few hours