UIPEthernet/UIPEthernet #157: Ethernet Issue

Mouli95, 2022-05-09:
Hardware used:
Atmel Atmega 4809 - 48 pin
ENC28j60 Ethernet module

[image]
Scenario 1 - Ethernet directly connected to a windows PC
Scenario 2 - Ethernet connected to a Router network

The Ethernet works fine in Scenario 1. In scenario 2 the ethernet doesn't work. I have checked the pining status and it is not pinging (I have attached the screenshot of the source code).

JAndrassy, 2022-05-09:
is the static IP valid for the network?
how it doesn't work?

Mouli95, 2022-05-09:
Yes, the static IP valid for network.

Mouli95, 2022-05-10:
how to disable dhcp in uipethernet library? @JAndrassy

JAndrassy, 2022-05-10:
same as in Ethernet. set static IP.
https://www.arduino.cc/reference/en/libraries/ethernet/ethernet.begin/

JAndrassy, 2022-05-10:
is the CS pin set right?

JAndrassy, 2022-05-10:
just to be sure. add Ethernet.init with a CS pin number

Mouli95, 2022-05-11:
I'm already try that also, Ethernet.init(7),
 Here I'm using Atmega4809 controller, 
SPI Pins :
[image]
   @JAndrassy 
MOSI - A4
MISO - A5
SCK - A6
CS - A7

JAndrassy, 2022-05-11:
you should use the EthernetENC library

Mouli95, 2022-05-11:
I'm tried EthernetENC library also. @JAndrassy