UIPEthernet/UIPEthernet #158: single port multi connection

snnzdmr, 2022-06-26:
Hello ,
I have 4 ports in total.

EthernetServer server = _server_1(1000);
EthernetServer server = _server_2(1001);
EthernetServer server = _server_3(1002);
EthernetServer server = _server_4(1003);

If I connect to the same port with 2 different clients, uipethernet accepts both connections.
Is there any way to prevent this ?
Only 1 client should be connected to each port

note: uipethernet lib 2.0.10

JAndrassy, 2022-06-27:
use server.accept(), not server.available()
https://www.arduino.cc/reference/en/libraries/ethernet/server.accept/

closed 2022-06-27 by JAndrassy

snnzdmr, 2022-06-27:
yes I am using "accept()".
I just stop the code after "server_1.begin()" is done
but even if I don't call the "accept()" function in the code, the same port accepts both connections.
[image]

JAndrassy, 2022-06-27:
you have to accept() and then stop()

snnzdmr, 2022-06-27:
ok.thank you for your help.