UIPEthernet/UIPEthernet #160: Getting localIP() and gatewayIP() even if ethernet removed from device

mightChamp, 2022-11-18:
Hello,

Device is connected to router since long time, and all works ok.
After removing Ethernet Cable from device, we get localIP, and Gateway IP by calling functions 
localIP() and gatewayIP() respectively.

and also Ethernet.mainatain() returns DHCP_CHECK_NONE;

Our purpose is to detect connection lost from router. 
So, How we get localIP() and gatewayIP(), if ethernet is not connected?

Is there anyway to check, device is connected to router?

Thank You

JAndrassy, 2022-11-18:
use `Ethernet.linkStatus()`

mightChamp, 2022-11-19:
Ethernet.linkStatus() will tell only is that Ethernet cable is connected or not, but doesn't update IP Address to 0.0.0.0.
Our purpose is to detect that device is connected to router or not. After sometime say(20-24 hours) device is not responding to ping. 
We continuously call Ethernet.maintian() function in loop.
We want to self diagnosis in device that it is connected in network or not, So, if it fails we can reinit Ethernet functions.

Thank You

JAndrassy, 2022-11-19:
the linkStatus is just a state of the connector, not the state of the network.
the only way to detect if some device is on the network is to try to contact it. 
even if you disconnect the cable from the network adapter, you still can have other network interface and even if if all network interfaces loose connection, you still have at least the loopback interface (127.0.0.1) so the network is up.

closed 2022-11-19 by JAndrassy