UIPEthernet/UIPEthernet #161: Arp padding

snnzdmr, 2022-12-15:
Hi, 
If I try to establish a TCP connection between ENC(TCP server) and PLC(TCP client), the PLC starts a "broadcast". There is padding inside.(I'm sniffing with wireshark)
ENC does not add padding when replying. So the PLC does not recognize the mac address of the ENC.
Can I add padding to the ENC reply?

[image]
[image]

snnzdmr, 2022-12-16:
[image]
[image]
[image]
[image]
  
  https://forum.wiznet.io/t/topic/2841 here it is said that the error is related to "mac address" but I tried again, same result.(iam using enc28j60) can you please help

JAndrassy, 2022-12-16:
I don't understand. padding is required for packets smaller than 60 bytes to add bytes up to 60 bytes. the enc28j60 adds the required padding,, not the library.

snnzdmr, 2022-12-17:
micro-St_f3:de:4b -> benim bilgisayarım(wireshark için)
Espessi_d1:29:58 -> ENC28j60
SiemensI_bb:13:6b -> Siemens PLC

I'm establishing a tcp connection to ENJ28j60(TCP server) with the computer(TCP client). Then my computerARP broadcasts and wants to learn the Mac address of ENC28j60.  ENJ28j60 is doing arp reply. My computer learns the mac address of enc28j60. no problem here. But

Siemens PLC(TCP Client) wants to connect ENC28j60(TCP server) . TCP connection is provided. Then Siemens PLC ARP broadcasts and wants to learn the Mac address of ENC28j60. But ENC28j60 does not reply to ARP. Then Siemens PLC closes the TCP connection because it cannot add the ENC28J60 to the ARP table.
You are right, when I look at the datasheet of ENC28j60, I see that the padding part is added as hardware (PADCFG2:PADCFG1)
But when I look at the wireshark screenshots, the only difference I see between the two broadcast requests is padding. There is padding in the arp request in Siemens PLC. There is no padding in the computer's arp request.

My whole problem is that enc28j60 reply to computer's broadcast. But not reply to siemens PLC's broadcast.
Why is enc28j60 not reply to arp requests with padding in it?
thank you for your help

JAndrassy, 2022-12-17:
why has the PLC such a prominent IP address? 192.168.0.1 it is usually the IP address of the device managing the network?

snnzdmr, 2022-12-17:
I don't know why plc's ip  address is 192.168.0.1. This is determined by the customer.
but I changed ito plc ip address 192.168.0.11 and tried it. again enc did not reply. I will share network device connection schema

snnzdmr, 2022-12-17:
Network 

[image]