UIPEthernet/UIPEthernet #164: Arduino Uno R4 mnima

shpaq22, 2024-01-21:
Hello, 

I'm using this library for Arduino Uno R3 with the ENC28J60 Ethernet module, and it works very well. 

Now, I've created a new setup with the latest hardware, using Arduino Uno R4 minima. Unfortunately, it doesn't work, and I'm missing declarations for SPI_MOSI, SPI_MISO, SPI_SCK, and SS.

Can you provide guidance on how to add these declarations to the library?

=======================================
#error "Not defined SPI_MOSI!"
#error "Not defined SPI_MISO!"
#error "Not defined SPI_SCK!"

JAndrassy, 2024-01-21:
use the EthernetENC library

shpaq22, 2024-01-21:
Thanks a lot! It's working :)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Nie
zawiera wirusów.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2024-01-22 by JAndrassy