UIPEthernet/UIPEthernet #167: client.conncted is false

GirishPawar1999, 2024-06-25:
after the connection the , code fetches the server and get the success code 200 too , but later one it enters the void loop  the "client.connected()" shows false, i.e not connected ... but if i remove the code of if(!client.connected()), the server if fetched.

I cant get this strange behavior, i wanted to add debug if the connection is lost by the server the system should not get stucked.

Microcontroller: ESP32 dev kit

#include <SPI.h>
#include <EthernetENC.h>
#include <ArduinoHttpClient.h>
//#include "utility/logging.h"

// Network settings
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
#define MYIPADDR 192,168,131,123
#define MYIPMASK 255,255,255,0
#define MYDNS 192,168,131,1
#define MYGW 192,168,131,1

// Server settings
char serverAddress[] = "60.254.18.10";
int port = 5000;

// Ethernet and HTTP client objects
EthernetClient ethernetClient;
HttpClient client(ethernetClient, serverAddress, port);

// Variables to measure the speed
unsigned long beginMicros, endMicros;
unsigned long byteCount = 0;

void setup() {
    Serial.begin(115200);
    delay(1000);
    Serial.println("Begin Ethernet");

    Ethernet.init(2);   // Initialize Ethernet with CS pin 2

    if (Ethernet.begin(mac)) { // Dynamic IP setup
        Serial.println("DHCP OK!");
    } else {
        Serial.println("Failed to configure Ethernet using DHCP");

        if (Ethernet.hardwareStatus() == EthernetNoHardware) {
            Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
            while (true) {
                delay(1); // do nothing, no point running without Ethernet hardware
            }
        }
        if (Ethernet.linkStatus() == LinkOFF) {
            Serial.println("Ethernet cable is not connected.");
        }

        IPAddress ip(MYIPADDR);
        IPAddress dns(MYDNS);
        IPAddress gw(MYGW);
        IPAddress sn(MYIPMASK);
        Ethernet.begin(mac, ip, dns, gw, sn);
        Serial.println("STATIC OK!");
    }
    delay(5000);

    Serial.print("Local IP : ");
    Serial.println(Ethernet.localIP());
    Serial.print("Subnet Mask : ");
    Serial.println(Ethernet.subnetMask());
    Serial.print("Gateway IP : ");
    Serial.println(Ethernet.gatewayIP());
    Serial.print("DNS Server : ");
    Serial.println(Ethernet.dnsServerIP());

    Serial.println("Ethernet Successfully Initialized");

    // Perform initial HTTP request
    performHttpRequest();
    beginMicros = micros();
}

void performHttpRequest() {
    Serial.println("Performing HTTP GET request...");
    client.get("/viewClients");

    int statusCode = client.responseStatusCode();
    String response = client.responseBody();
    
    Serial.print("Status code: ");
    Serial.println(statusCode);
    Serial.print("Response: ");
    Serial.println(response);

    if (statusCode == 200) {
        byteCount += response.length();
    } else {
        Serial.println("HTTP request failed.");
    }
}

void loop() {
    if (Ethernet.linkStatus() == LinkON) {
        if (!client.available()) {
            Serial.println("Client is not connected. Reconnecting...");
            ethernetClient.stop();
            delay(1000);
            // Ensure the HttpClient is properly reinitialized
            client.stop();
            client = HttpClient(ethernetClient, serverAddress, port);
            delay(1000);
            Serial.println("Client reconnected.");
        }

        Serial.println("Client connected. Making GET request...");
        performHttpRequest();
    } else {
        Serial.println("Ethernet link is down. Reconnecting...");
        Ethernet.begin(mac);
        delay(2000);
    }

    delay(1000);  // Wait 10 seconds before the next loop iteration
    Ethernet.maintain();
}

log ====
5:33:18.068 -> Status code: 200
15:33:18.068 -> Response: ["M983-01-I"]
15:33:19.095 -> Client is not connected. Reconnecting...
15:33:21.063 -> Client reconnected.
15:33:21.110 -> Client connected. Making GET request...
15:33:21.110 -> Performing HTTP GET request...
15:33:22.289 -> Status code: 200
15:33:22.289 -> Response: ["M983-01-I"]
15:33:23.278 -> Client is not connected. Reconnecting...
15:33:25.313 -> Client reconnected.
15:33:25.313 -> Client connected. Making GET request...
15:33:25.313 -> Performing HTTP GET request...
15:33:27.539 -> Status code: 200
15:33:27.539 -> Response: ["M983-01-I"]
15:33:28.530 -> Client is not connected. Reconnecting...

JAndrassy, 2024-06-25:
```
if (!client.available()) {
      Serial.println("Client is not connected. Reconnecting...");
```
`available` returns the count of bytes available for read, not the connection status

GirishPawar1999, 2024-06-25:
later on i changed it to 

if(!client. connected()){
  Serial.println("Not Connected");
}

it still moves under the same

GirishPawar1999, 2024-06-25:
"#include <SPI.h>
#include <EthernetENC.h>
#include <ArduinoHttpClient.h>
//#include "utility/logging.h"

// Network settings
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
#define MYIPADDR 192,168,131,123
#define MYIPMASK 255,255,255,0
#define MYDNS 192,168,131,1
#define MYGW 192,168,131,1

// Server settings
char serverAddress[] = "60.254.18.10";
int port = 5000;

// Ethernet and HTTP client objects
EthernetClient ethernetClient;
HttpClient client(ethernetClient, serverAddress, port);

// Variables to measure the speed
unsigned long beginMicros, endMicros;
unsigned long byteCount = 0;

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("Begin Ethernet");

  Ethernet.init(2);   // Initialize Ethernet with CS pin 2

  if (Ethernet.begin(mac)) { // Dynamic IP setup
    Serial.println("DHCP OK!");
  } else {
    Serial.println("Failed to configure Ethernet using DHCP");

    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
      }
    }
    if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }

    IPAddress ip(MYIPADDR);
    IPAddress dns(MYDNS);
    IPAddress gw(MYGW);
    IPAddress sn(MYIPMASK);
    Ethernet.begin(mac, ip, dns, gw, sn);
    Serial.println("STATIC OK!");
  }
  delay(5000);

  Serial.print("Local IP : ");
  Serial.println(Ethernet.localIP());
  Serial.print("Subnet Mask : ");
  Serial.println(Ethernet.subnetMask());
  Serial.print("Gateway IP : ");
  Serial.println(Ethernet.gatewayIP());
  Serial.print("DNS Server : ");
  Serial.println(Ethernet.dnsServerIP());

  Serial.println("Ethernet Successfully Initialized");

  // Perform initial HTTP request
  performHttpRequest();
  beginMicros = micros();
}

void performHttpRequest() {
  Serial.println("Performing HTTP GET request...");
  client.get("/viewClients");

  int statusCode = client.responseStatusCode();
  String response = client.responseBody();

  Serial.print("Status code: ");
  Serial.println(statusCode);
  Serial.print("Response: ");
  Serial.println(response);

  if (statusCode == 200) {
    byteCount += response.length();
  } else {
    Serial.println("HTTP request failed.");
  }
}

void loop() {
  if (Ethernet.linkStatus() == LinkON) {
    Serial.println(client.connected());
    if (!client.connected()) {
      Serial.println("Client is not connected. Reconnecting...");
      ethernetClient.stop();
      delay(1000);
      // Ensure the HttpClient is properly reinitialized
      client.stop();
      client = HttpClient(ethernetClient, serverAddress, port);
      delay(1000);
      Serial.println("Client reconnected.");
    }

    Serial.println("Client connected. Making GET request...");
    performHttpRequest();
  } else {
    Serial.println("Ethernet link is down. Reconnecting...");
    Ethernet.begin(mac);
    delay(2000);
  }

  delay(1000);  // Wait 10 seconds before the next loop iteration
  Ethernet.maintain();
}
"

above code follows the observation :
17:43:28.362 -> Client is not connected. Reconnecting...
17:43:30.367 -> Client reconnected.
17:43:30.418 -> Client connected. Making GET request...
17:43:30.418 -> Performing HTTP GET request...
17:43:31.630 -> Status code: 200
17:43:31.630 -> Response: ["M983-01-I","M983-01-B"]
17:43:32.608 -> 0
17:43:32.608 -> Client is not connected. Reconnecting...
17:43:34.613 -> Client reconnected.
17:43:34.613 -> Client connected. Making GET request...
17:43:34.613 -> Performing HTTP GET request...
17:43:35.831 -> Status code: 200
17:43:35.831 -> Response: ["M983-01-I","M983-01-B"]
17:43:36.857 -> 0
17:43:36.857 -> Client is not connected. Reconnecting...
17:43:38.867 -> Client reconnected.
17:43:38.867 -> Client connected. Making GET request...
17:43:38.867 -> Performing HTTP GET request...
17:43:40.080 -> Status code: 200
17:43:40.080 -> Response: ["M983-01-I","M983-01-B"]
17:43:41.059 -> 0
17:43:41.059 -> Client is not connected. Reconnecting...

JAndrassy, 2024-06-25:
the ArduinoHttpClient library sends in HTTP header  "Connection: close" so the server closes the connection

GirishPawar1999, 2024-06-25:
How to resolve this ? to know if the connection status

JAndrassy, 2024-06-25:
`client.connectionKeepAlive();`

https://buymeacoffee.com/jurajandray